LongSoft/UEFITool #258: Could this software be a UEFI replacement for Universal BIOS Backup Toolkit? If not, in the future?

Unknown78, 2022-03-09:
Universal BIOS Backup Toolkit official thread (need account):
https://forums.mydigitallife.net/threads/universal-bios-backup-toolkit.9856/
```
Download links from the thread:
https://www.mirrored.to/files/5VA4OF4I/Universal_BIOS_Backup_ToolKit_2.0.zip_links
https://mega.nz/#!ElJGFRDA!vmwxdI9w6cjgtm87v7rJn7cWTcza19jDec25bix-K0E

  File: Universal BIOS Backup ToolKit 2.0.zip
CRC-32: 3ffd16fe
   MD4: 9debd27450f1efa7496fb1ca23eb58ed
   MD5: 3521401d8865c20f52c7098b32e6b6e0
 SHA-1: 3e3aadb80afe04eab1574b92dd25317ea1ea87b1

  File: Universal BIOS Backup ToolKit 2.0.exe
CRC-32: 96846278
   MD4: a5889579c24efdb52350d69a8e1ec88a
   MD5: e20002dba2f2129797400b4bbf5cace7
 SHA-1: 0bed0f6320cf705ab454ce14decc0a3fc6840337
```
[image]

Universal BIOS Backup Toolkit by DavidXXW has the ability to backup BIOS firmware, but not for UEFI.

Will this software could be replacement? Any plan? If not, in the future?

NikolajSchlej, 2022-03-09:
No, this software is for parsing and editing an already created dump of the firmware, not for creating said dump. However, there are both OSS and proprietary projects that can replace UBBT: [flashrom](https://www.flashrom.org/Flashrom), [Intel Flash Programming Tool](https://www.win-raid.com/t596f39-Intel-Converged-Security-Management-Engine-Drivers-Firmware-and-Tools.html), and so on.

closed 2022-03-09 by NikolajSchlej

modz2014, 2024-05-10:
it backs up BIOS