LongSoft/UEFITool #260: IFWI 2.0 BIOS

PaulNevada, 2022-03-25:
Can A59 be used for opening the file for saving the rom version only? I want to use it later to rollback my laptop's firmware...

NikolajSchlej, 2022-03-25:
Yes, but that's not a good way to perform a rollback, because all individual board data that were present in the original image will be lost if you flash that "ROM" image somehow.
You can try using a compatible version of Intel Flash Programming Tool, or a hardware SPI flashing device to save and restore a full backup of SPI flash on your laptop.

closed 2022-03-25 by NikolajSchlej

PaulNevada, 2022-03-25:
Do you mean I could receive a brick this way? And if I will rollback so and then update the regular way with on of the next official file - won't all the potential lost bios data come back to its origianl place?

NikolajSchlej, 2022-03-25:
IDK about bricking, but you could loose your board data this way. A factory update file, even extracted from it's Capsule, is not a full backup, because it lacks the individual board data. A better backup would be an image of your current BIOS made using Intel FPT or HW flash programmer.

PaulNevada, 2022-03-25:
Not an update file, but a 8MB-in-size file. Isn't it a full one?

PaulNevada, 2022-03-25:
What is that "individual board data"?

NikolajSchlej, 2022-03-28:
Individual board data is motherboard serial number, motherboard UUID and some other things.
[image]

PaulNevada, 2022-03-28:
Thanks for the given info!