LongSoft/UEFITool #262: [Question] Is there any way to know if the Bios image has AMD PSB support ?

AkechiShiro, 2022-04-06:
Hi,

I'm looking at different bios updates from Lenovo specifically, and there was a recent polemic around Lenovo applying AMD PSB to some of their machine, essentially vendor locking the CPU to only their motherboards. I want to know if there is a way to check if there is such a feature by looking into the BIOS firmware.

Just like there is a way to check for Intel BootGuard support (which is currently possible using UEFITool) is there any way AMD PSB (Platform Secure Boot) can be detected in any way ?

I'm also trying to find when AMD PSB first landed in Ryzen's CPUs, is it present in all CPUs ? I think the feature can only be enabled by the vendor, if so how can we make sure this feature is indeed enabled (like should we just brick the machine intentionally flashing a custom BIOS ?)

If there is no way at the moment, could there be work for a way to do this check ? How could we get started, I'm willing to give a try at working at this.

NikolajSchlej, 2022-08-24:
I think some other projects like https://github.com/PSPReverse/PSPTool might answer this question. UEFITool had been developed before PSP got popular, and neither me nor @vit9696 had an AMD machine or a need to support them in UEFITool since.

If you would like to try adding such a support, consider learning how PSPTool works and porting the structures and parsers from there enough to be able to get to UEFI firmware volumes.

closed 2022-08-24 by NikolajSchlej

AkechiShiro, 2022-08-24:
A recent implementation for this has landed ? Or you're still recommanding to use other tools ?

NikolajSchlej, 2022-08-24:
I still recommend using PSPTool, because UEFITool currently has little to no support for AMD-specific things like PSP structures, and will unlikely to get it any time soon, if somebody won't contribute a bunch of code to implement it. 

I'll create another issue about adding support for AMD-specific things.

AkechiShiro, 2022-08-24:
Be sure to tag me on the new issue, I might try to help or test