LongSoft/UEFITool #269: Support parsing of AMD-specific image formats and data structures

NikolajSchlej, 2022-08-24:
We already support a lot of Intel-specific things in UEFITool, mostly because back in 2013-2015 AMD CPUs weren't as popular as they are today (in 2022 that is). Since then AMD started using PSP and put their UEFI firmware volumes into PSP-related data structures, "directories", whatever else.

Fiano has some support for it, and the structures already are reverse engineered by the good folks of PSPTool project, so there are multiple sources to be inspired by (and yoink some definitions from).

NikolajSchlej, 2022-08-24:
CC @AkechiShiro

NikolajSchlej, 2025-03-22:
CC @PatrickRudolph, who already started this work on his fork here: https://github.com/PatrickRudolph/UEFITool/commits?author=PatrickRudolph

NikolajSchlej, 2025-03-22:
@PatrickRudolph, will you be fine with me borrowing your commits into the mainline? This work is long overdue (back in 2015 when I last had time for active UEFITool development, AMD CPUs were much less popular than today).

PatrickRudolph, 2025-03-22:
@NikolajSchlej The goal is to add proper AMD support in upstream. I found the task quite challenging as the directory structure is completely different compared to existing Intel images and is also inconsistent across generations.
I'll push my latest changes as reference which work quite well but have lots of potential for improvements.

PatrickRudolph, 2025-03-22:
I've pushed my latested changes here:
https://github.com/PatrickRudolph/UEFITool/commits/amd_parser

It still has duplicated BIOS region entries as different L2 directory tables point to the same spot in flash, but it's not properly detected.

NikolajSchlej, 2025-04-30:
Integrated the code written by @PatrickRudolph as https://github.com/LongSoft/UEFITool/commit/15b2d53747d95ff1f44f6da4afd74b9e6f292b7c, will iterate on improving it, hope to release in A72.