LongSoft/UEFITool #276: Invisible text on macOS 12 with dark mode

vit9696, 2022-08-27:
Apparently there is white text on white background in some places:

[image]
[image]

NikolajSchlej, 2022-09-10:
Can be resolved by switching to Qt 6.x for GH-made releases, with a 5.6.3 builds to remain supported.

NikolajSchlej, 2022-09-10:
I've tested on Qt 6.3.1, looks fine enough, but requires some minor changes to make protected range marking less glaring in dark mode.

NikolajSchlej, 2022-09-10:
[image]

yeggor, 2022-09-10:
I just now noticed that this has already been discussed. Do you think this solves the problem with protected ranges colors: https://github.com/LongSoft/UEFITool/pull/293?

NikolajSchlej, 2022-09-13:
Support for macOS dark mode had been recently added to Qt6, but had not been backported to Qt5 in any way, so we should start using Qt6 (but have remain compatible with the last Qt build under LGPLv2.1 - 5.6.3, with some minor UI issues if needed) and get that for free.
- [ ] switch to static Qt6.x for macOS and Windows
- [ ] provide a fallback to static Qt 5.6.3 used now
- [ ] add that fallback to CI/CD pipeline to not break it by random changes

@yeggor also showed that darkX colors are much nicer-looking in dark mode in #293, but I'd like to detect both Aqua and Dark modes, and switching between them dynamically.
- [ ] detect the current mode dynamically, adjust BootGuard marking colors to suit the current mode

Windows 11 also implemented Dark Mode support, so it also needs to be supported the same way.
- [ ] make the above work in Windows 11

Some Linux DE/WM also support dark mode, but they are way too many of them for us to write any DE/WM-specific code that Qt6 doesn't cover.

NikolajSchlej, 2023-02-08:
Looks like we've also lost Windows XP support even for 5.6.3-based builds, this needs to be fixed by generating a separate win32 executable with 'vc141_xp' target (CMake doesn't support Qt 5 for real, so that needs to be done by pre-generated make files)
- [ ] Make sure Windows XP remains supported after all the changes, because people still run the tool there after all this years

NikolajSchlej, 2023-02-11:
Update: Qt 6.4.2 as of today does not support dark mode for its default style.

NikolajSchlej, 2023-02-11:
[Detection of dark modes enabled on macOS and Windows](https://successfulsoftware.net/2021/03/31/how-to-add-a-dark-theme-to-your-qt-application/).

NikolajSchlej, 2023-02-12:
The new win64 build of UEFITool uses Qt 6.4.2 that supports experimental dark mode by switching to classic widgets with a darker palette. The application does not start in this mode by default, but can be forced into it by adding `-platform windows:darkmode=2` commandline parameter to UEFITool.exe invocation. 

Looks like this:
[image]

NikolajSchlej, 2023-02-12:
Got Windows 10+ dark mode working as expected, no overrides required:
[image]

NikolajSchlej, 2023-02-12:
Now detection of dark mode and setting colors also works in macOS:
[image]

NikolajSchlej, 2023-02-12:
Done, the current CI/CD pipeline can successfully generate Qt-6.4.2-based builds for both macOS and Windows, and uses a native installed Qt5/6 on Linux. I consider this issue to be resolved, please open another one if UI Dark Mode doesn't work for you in some way.

closed 2023-02-12 by NikolajSchlej

NikolajSchlej, 2023-02-12:
Confirmed macOS way to determine UI dark mode to work in Linux, enabled it there.
[image]