LongSoft/UEFITool #300: cmake config does not generate an install target

vulpes2, 2022-09-25:
The cmake config provided in the repo does not generate an install target, so binaries have to be copied manually after they've been built. It would be nice if there is an install target which copies the binaries and .desktop file + icons (only applicable for UEFITool) to the system. I believe this is the corresponding documentation for generating this target: https://cmake.org/cmake/help/latest/command/install.html

```
$ mkdir build && cd build
$ cmake ..
$ make -j$(nproc)
$ make install 
make: *** No rule to make target 'install'.  Stop.
```

klemensn, 2022-11-27:
https://github.com/LongSoft/UEFITool/blob/47c8938c7eb51d270ed6366bfc55f27670ce957e/UEFITool/uefitool.desktop#L6-L8
BSDs default to `/usr/local` and the current executable is called `UEFITool`, so these need to be taken care of either way before the .desktop file is being shipped, I think.

The icon/png files seem to be lowercased already in-tree, so that should work.

NikolajSchlej, 2023-02-12:
Need some help here in understanding the minimal set of requirement for every major OS target we support (Windows, macOS, Linux, FreeBSD) and setting all the things correctly.
I don't want to rename `UEFITool` to lowercase, but I'm completely fine to update that desktop file if it's not correct (probably as a build step, because the default paths are different between OSes).

NikolajSchlej, 2023-02-12:
This is how @vulpes2's FlatPak generator YAML does installing:
```
- mkdir -p ${FLATPAK_DEST}/{bin,share/{applications,appdata,icons/hicolor/512x512/apps}}
- cp UEFITool/icons/uefitool_512x512.png ${FLATPAK_DEST}/share/icons/hicolor/512x512/apps/uefitool.png
- cp UEFITool/uefitool.desktop ${FLATPAK_DEST}/share/applications
- cp appstream/appdata.xml ${FLATPAK_DEST}/share/appdata/${FLATPAK_ID}.appdata.xml
- cp build/UEFIExtract/UEFIExtract ${FLATPAK_DEST}/bin/uefiextract
- cp build/UEFIFind/UEFIFind ${FLATPAK_DEST}/bin/uefifind
- cp build/UEFITool/UEFITool ${FLATPAK_DEST}/bin/uefitool
```
Should work for Linux, with some adjustments also for FreeBDS and macOS regarding the `/usr/local` paths.
Windows can be ignored for this install script, because it's unlikely anybody will use it there anyway.

NikolajSchlej, 2023-02-12:
Added the targets in https://github.com/LongSoft/UEFITool/commit/4053a8fe721b676428c07e6a985a120f43e66504
Need to install the icon on Linux and FreeBSD and that will be it.

NikolajSchlej, 2023-02-12:
Should be fixed in https://github.com/LongSoft/UEFITool/commit/a5a0f55149b6f1531bac401d5f57f5939af1d1f1
Will close this one as complete once I can confirm that both Linux and FreeBSD targets work properly.

NikolajSchlej, 2023-02-12:
Confirmed working in Fedora 37, closing as complete.

closed 2023-02-12 by NikolajSchlej