LongSoft/UEFITool #301: compilation error

llxiaoyuan, 2022-09-27:
UEFITool-new_engine\common\kaitai\kaitaistream.cpp:32:20: fatal error: endian.h: No such file or directory
 #include <endian.h>
                    ^
compilation terminated.
Makefile.release:4215: recipe for target 'release/kaitaistream.o' failed
mingw32-make: *** [release/kaitaistream.o] Error 1

Cleaning up...

Done!

NikolajSchlej, 2022-09-27:
Reporting like this is not actionable, so for us to be able to fix it, please provide:
- your OS and MinGW versions
- steps to reproduce the issue

So far MinGW is a secondary target, default compiler for Windows is MSVC2019. #282 tracks adding more compilers to CI/CD pipeline.

NikolajSchlej, 2022-09-27:
Also confirmed that MinGW 8.1.0 has a bug that prevents usage of <filesystem> C++ std header, and Qt 6.x uses it extensively, so using Qt 6.x builds require newer MinGW versions (CI/CD pipeline currently runs with 11.2.0).

NikolajSchlej, 2022-09-27:
Fixed.

PR #303 (2022-09-27): Fix building with MinGW 11.2.0 on Windows, add it into CI/CD
Fixes #301.
closed 2022-09-27 by NikolajSchlej