LongSoft/UEFITool #307: Extracting ACPI tables from firmware images

vulpes2, 2022-10-04:
Feel free to close this if you think it's out of scope, but it would be really nice if UEFITool could extract ACPI tables from an image. I'm aware that using `acpidump` and `acpixtract` on a running machine is probably the easiest way to obtain them, but having an option that doesn't require ownership of the physical device seems useful too.

vulpes2, 2022-10-04:
I checked some AMI images I have, and DSDT seems to be stored in `9F3A0016-AE55-4288-829D-D55FD3AAC347`, named `AmiBoardInfo` or `AmiBoardInfo2`.

NikolajSchlej, 2022-10-04:
I don't think we need any special handling for ACPI tables, because they can already be extracted using UEFIExtract on the whole image, then filtering for the files with relevant 4CC codes (```FADT```, ```MADT```, ```DSDT```, ```SSDT```, etc.).

closed 2022-10-04 by NikolajSchlej