LongSoft/UEFITool #309: Save Key Manifest Public Key as an artifact

hughsie, 2022-10-09:
At the moment UEFITool just prints the Boot Policy Key Signature like this on stdout:

	Boot Policy Key Signature:
	Version: 10h
	KeyId: 0001h
	SigScheme: 0016h
	Boot Policy Public Key (Exponent: 10001h): 

This makes it hard for the LVFS to parse it. Ideally we'd save it in the _dump directory_ as an artefact that we can just load as an asset like we do all the other extracted information. This would let us load the pubkey on the LVFS like we do other assets, like this:

[image]

This would then allow us to check that the Boot Policy Key isn't one that's been accidentally leaked.

NikolajSchlej, 2022-10-09:
Needs to do the same with everything mentioned on Security page, i.e. not only KeyManifest, but also BootPolicy and ACMs.

NikolajSchlej, 2022-10-09:
Meanwhile, I've added both SHA256 and SHA386 of KM pubkey to Security page (and UEFIExtract stdout, by that).