LongSoft/UEFITool #311: terminate called after throwing an instance of 'std::out_of_range'

hughsie, 2022-10-11:
Using tip of new_engine:

	$ ./UEFIExtract/UEFIExtract firmware.bin
	terminate called after throwing an instance of 'std::out_of_range'
	  what():  basic_string::substr: __pos (which is 18446744073707980765) > this->size() (which is 4458461)
	Aborted (core dumped)

...which explodes here:

	#8  0x00007ffff7ca559f in std::__throw_out_of_range_fmt (__fmt=__fmt@entry=0x7ffff7dae348 "%s: __pos (which is %zu) > this->size() (which is %zu)")
	    at ../../../../../libstdc++-v3/src/c++11/functexcept.cc:101
	#9  0x00007ffff7d4dca4 in std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_check (__s=0x7ffff7db2c5a "basic_string::substr", 
	    __pos=<optimized out>, this=<optimized out>)
	    at /usr/src/debug/gcc-12.2.1-2.fc37.x86_64/obj-x86_64-redhat-linux/x86_64-redhat-linux/libstdc++-v3/include/bits/basic_string.h:382
	#10 std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::substr (this=<optimized out>, __pos=<optimized out>, __n=<optimized out>)
	    at /usr/src/debug/gcc-12.2.1-2.fc37.x86_64/obj-x86_64-redhat-linux/x86_64-redhat-linux/libstdc++-v3/include/bits/basic_string.h:3124
	#11 0x000000000044bf01 in UByteArray::mid (this=0x7fffffffd518, pos=-1570851, len=524288) at ../common/bstrlib/../ubytearray.h:68
	#12 0x0000000000441be3 in FfsParser::checkProtectedRanges (this=0x7fffffffd4e0, index=...) at ../common/ffsparser.cpp:3645
	#13 0x000000000043efbd in FfsParser::performSecondPass (this=0x7fffffffd4e0, index=...) at ../common/ffsparser.cpp:3308
	#14 0x000000000042b0ae in FfsParser::parse (this=0x7fffffffd4e0, buffer=...) at ../common/ffsparser.cpp:132

The firmware is private, but it's from the Surface Laptop 2 v34.106.75.0 -- which I'm sure can be found online.

NikolajSchlej, 2022-10-11:
Forgotten size check once again, will fix when I'm back from a trip.

NikolajSchlej, 2022-10-11:
Microsoft decided to put the volumes inside an update capsule in a random order, which is spec-compliant, but still really unfortunate, because computing physical addresses by assuming the last VTF ends at 0xFFFFFFFFF, and everything else is directly above it is impossible in this configuration, and yields incorrect results.

NikolajSchlej, 2022-10-11:
```
./UEFIExtract SL2.bin 
libc++abi: terminating with uncaught exception of type std::out_of_range: basic_string
zsh: abort      ./UEFIExtract SL2.bin
```
Reproduced on the attached file: [SL2.bin.zip](https://github.com/LongSoft/UEFITool/files/9752145/SL2.bin.zip)

NikolajSchlej, 2022-10-11:
Fixed in c3cedba150f05f1d56b41e2348151687cacb3384

closed 2022-10-11 by NikolajSchlej

hughsie, 2022-10-11:
Agree, thanks!