LongSoft/UEFITool #312: terminate called after throwing an instance of 'std::out_of_range'

hughsie, 2022-10-11:
Processing specifically Fujitsu firmware I get:

	terminate called after throwing an instance of 'std::out_of_range'
	  what():  basic_string::substr: __pos (which is 19922944) > this->size() (which is 4096)

this is what gdb says:

	#8  0x00007ffff7ca559f in std::__throw_out_of_range_fmt (__fmt=__fmt@entry=0x7ffff7dae348 "%s: __pos (which is %zu) > this->size() (which is %zu)")
	    at ../../../../../libstdc++-v3/src/c++11/functexcept.cc:101
	#9  0x00007ffff7d4dca4 in std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_check (__s=0x7ffff7db2c5a "basic_string::substr", 
	    __pos=<optimized out>, this=<optimized out>)
	    at /usr/src/debug/gcc-12.2.1-2.fc37.x86_64/obj-x86_64-redhat-linux/x86_64-redhat-linux/libstdc++-v3/include/bits/basic_string.h:382
	#10 std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::substr (this=<optimized out>, __pos=<optimized out>, __n=<optimized out>)
	    at /usr/src/debug/gcc-12.2.1-2.fc37.x86_64/obj-x86_64-redhat-linux/x86_64-redhat-linux/libstdc++-v3/include/bits/basic_string.h:3124
	#11 0x000000000044bebf in UByteArray::mid (this=0x7fffffffd470, pos=19922944, len=13631488) at ../common/bstrlib/../ubytearray.h:68
	#12 0x000000000042cd2a in FfsParser::parseIntelImage (this=0x7fffffffd500, intelImage=..., localOffset=0, parent=..., index=...) at ../common/ffsparser.cpp:405
	#13 0x000000000042b1e5 in FfsParser::performFirstPass (this=0x7fffffffd500, buffer=..., index=...) at ../common/ffsparser.cpp:159
	#14 0x000000000042b06f in FfsParser::parse (this=0x7fffffffd500, buffer=...) at ../common/ffsparser.cpp:129

Reproducer here: https://people.freedesktop.org/~hughsient/temp/blob.bin

NikolajSchlej, 2022-10-11:
Attaching here. Will check this out when back, as coding while on a beach makes keyboards die.
[blob.bin.zip](https://github.com/LongSoft/UEFITool/files/9757545/blob.bin.zip)

NikolajSchlej, 2022-10-15:
https://github.com/LongSoft/UEFITool/commit/50396d729136e211ed6b282c9eb67691ecf4c22f fixes it for me, please test on real images and see if it helps.
Also, I'l like to get a HW model of that Fujutsu thing that triggers this, so I can find a dump and see what's really wrong with it, and maybe report it to Fujitsu, because flash descriptor containing phantom oversized regions is bug.

hughsie, 2022-10-15:
D3920-A1 R1.19.0, but check your DMs

NikolajSchlej, 2022-10-20:
Closing this as completed, please ping me if this is still reproducible on ToT.

closed 2022-10-20 by NikolajSchlej