LongSoft/UEFITool #314: Revese byte order in public key modulus

phretor, 2022-10-11:
I think the way each public key modulus is printed in the "Security" tab is reversed.

```
Intel BootGuard Key manifest found at base 1FFA280h
Tag: '__KEYM__'
Version: 21h
KmVersion: 01h
KmSvn: 00h
KmId: 01h
KeySignatureOffset: 0044h
FPFHashAlgorithmId: 000Ch
HashCount: 0001h
KM Hashes:
UsageFlags: 0000000000000001h, SHA256: D5502FF061699F9E2B7C64AB41374F56AE6F45DB870DDBA4733DDC30323878BB
Key Manifest Key Signature:
Version: 10h
KeyId: 0001h
SigScheme: 0016h
Key Manifest Public Key (Exponent: 10001h): 
53450FF591470C7DC3D7430B3F99780D3D6C87B54B32224D6A1CA88394C4C083
1393D9D6BD5F9C2C2EDF86959CAAA36917C05EF42B9E0FC16C966E0FDF968014
FE51A7886313EA804815D5ED249E7DFE177C518C023B280FCF62D713E04895FC
3BE67C5FDEE26FD3F502A60C8BE9CCADEFE632D53807EE7E67875905239BC88B
84C02FAA4E095C778E5087794EC6C36F39F950C8935CA62C4C2E3F2AC3025A9C
6D0D4E3E5B3CF85DC78DAFC7EAADCD2F1337FF4F8CF3FAE48BC7B631E464A923
DD1F602F17A22289C2F4EC5657289D801831EEC84CF8A79087E47A7AFBF3F085
AE42B054D4A3FD2C4F87827946EA4FA24581B7CD7633496E5DB89385A947E2C0
7B24EB7FD9BDA266F6ABE60789E1DEEE1A27C9CC33CB86E0A62AD8E5FCF2A0CC
52368BC73B2DC209CF2A6AD380E597DEC884051AF8883042675D95617A8B9902
446948E896C1A1375B64FBCE46F0B3927F77D1B1B6B3D6E0069DF307AFC7E6D6
0E113996038C348BBF0EA8DAB5025A61A140349957E8A00589A6E7B9CB5043E3
Key Manifest Signature: 
4F299E2575A552479FDB64B65C1C58FAFD7C1B533E181C3760F355393DB581B6
F04B0E9BB995F287CA469FA716D56F7630DB24975EFE3AE85C25CD53A0487190
15DB6C158F06C7521C86156FF8331601EFFB8838C134B655DC48E367BF38A449
79EB85AD9273DD16779E760A36CE5683E438ED037D54378DD6399F788618DD68
BAD0D96A6F0EA15B296C80DFC07A3FCFECC79379D072846A06A98BD5ED4D44DE
ED02A5A188D3B0431DCC555BB2DC144085F01A3B10B4DD61C6475DBF0862FB39
AA5C38ACF30DE0B32967235979D971323B1B2E957F89FC2C4FA969132824F92A
095D4A249A342A9BC2B98D8710E88476CC91239CBC689A93898E79988DA1F39D
F63D050792064CE6E407913B79C774D45CB00E3019A0B03903F6AF7C412E859D
72EB3C12D6A98437E7DA3F073FFC375B5429630E068D91F1377D28C8F52AB139
4B3707070E8BBE89A59F47373B711F7CB18C934D06E7687A150F271AC8B45844
3A42A9459DC7DD8E85698C360171A0696064CBCC2795789163FE4F09F692243B
```

Here, that `53450FF5914...A6E7B9CB5043E3` actually corresponds to to the reverse (1 byte granularity) of the public key modulus, while the UI of UEFITool names it as "Public Key".

See [this discussion](https://github.com/binarly-io/fwhunt/pull/7#issuecomment-1273950251) and [how patterns can be extracted](https://github.com/phretor/intel-leak-checker/#how-to-generate-the-patterens-from-private-keys) from public keys (i.e., by reversing the byte order of the modulus of the public key).

NikolajSchlej, 2022-10-11:
This are the raw bytes as they are found in the images, so it's PEM/DER things that are reversed, not this raw data. The hashes computed over KM modulus in this format are stored in FPFs, and the hashes computed over BP modulus in this format are stored in KM, so for UEFITool's standpoint, this format is "true", and everything else is "reversed".

closed 2022-10-11 by NikolajSchlej

phretor, 2022-10-11:
Okay, I understood the viewpoint and, for consistency, it makes a lot of sense of course. Thanks!