LongSoft/UEFITool #318: Add support to extract files from ME region

miczyg1, 2022-11-07:
ME region follows its own rules for its filesystem. It is entirely not related to UEFI spec. However, despite it is not UEFI-compliant, UEFITool can parse it with ease. In some cases it may be beneficial to extract single files from ME region, e.g. extracting PCIe 5.0 FW to integrate it in coreboot builds: https://review.coreboot.org/c/coreboot/+/68987

The goal of such a procedure is to avoid PCIe 5.0 disfunction when ME is disabled ,e.g. by setting HAP bit. PCIe 5.0 requires fetching the PCIe 5.0 FW from ME through MEI/HECI during BIOS POST, which will become impossible when ME is disabled. However, including the extracted binary allows to workaround the disabled ME problem and keep PCIe 5.0 functional.

Currently, the firmware extraction is possible via UEFITool GUI, however, it would be beneficial to have a scriptable method of extracting such blobs with UEFIExtract for automation purposes.

NikolajSchlej, 2022-11-07:
Indeed, my usual method would be to extract everything and then use ```find``` or ```bgrep```, but it's certainly not scalable.

- [ ] Add UEFIExtact console UI command to extract an item with a given 'Name' and/or 'Text'.
- [ ] Improve UEFIExtract usage text to fully explain the possible set of parameters.

miczyg1, 2022-11-07:
>  Add UEFIExtract console UI command to extract an item with a given 'Name' and/or 'Text'.

it could even be a path if it is nested... However, this would make the whole command too long. Names can sometimes be duplicated inside one image.