LongSoft/UEFITool #323: Label More GUIDs / Regions Within Intel Boot Guard Region

EmperorArthur, 2022-12-16:
Please Add the following GUIDs to the database.

* "C30FFF4A-10C6-4C0F-A454-FD319BAF6CE6": The body starts with "__ACBP__".  This is the "Bootguard Policy"
* "7C9A98F8-2B2B-4027-8F16-F7D277D58025": The body starts with "__KEYM__".  This is the "BootGuard Key Manifest"

----

Per my [understanding](https://github.com/LongSoft/UEFITool/wiki/Intel-Boot-Guard), yellow means something which is partly within the boot guard, and red means IBB protected.

Examining a Dell flash dump I see three interesting GUIDs within a FFSV2 Filesystem, with a Volume GUID of "61C0F511-A691-4F54-974F-B9A42172CE53".  Every other GUID within that filesystem is red, yet these three and the filesystem itself is yellow.  I suspect this is because of the padding not explicitly being checked.

* "6520F532-2A27-4195-B331-C0854683E0BA": Unknown
* "C30FFF4A-10C6-4C0F-A454-FD319BAF6CE6"
* "7C9A98F8-2B2B-4027-8F16-F7D277D58025"

Information taken from [this post](https://stackoverflow.com/a/61568389).

NikolajSchlej, 2022-12-20:
Yellow means "partially covered by any kind of protection known to UEFITool", the color scheme likely needs an update to remove ambiguity. 
As for GUIDs, those are specific to a single OEM or IBV, and Intel makes very little demands on where to put BootGuard structures in the image, namely:
- everything needs to be close enough to the end of the BIOS image (how close exactly depends on chipset generation, but generally in the last 4 Mb)
- none of the structures can themselves be covered by BootGuard hashes (to prevent chicken-and-egg issue and make parsing easier for the ACM). This is why you have those yellow "holes" in the sea of red, BTW.

NikolajSchlej, 2023-01-13:
Done in 1adff585916eac70b034adc2c2dc5a3520e13117

closed 2023-01-13 by NikolajSchlej