LongSoft/UEFITool #326: What get SPD info about RAM ?

selevo, 2022-12-30:
in more tablets ram memory chips solderign on board and no spd eeprom, it's info include UEFI bios chip.
What get and edit this infomation ?

NikolajSchlej, 2023-01-03:
Closing as a non-issue for UEFITool, but I can direct you to [this link](https://www.wikiwand.com/en/Serial_presence_detect#introduction), which explains how the SPD table should look like for a given generation of SDRAM. Convert the table values into binary strings and then search for them using the usual "Search" window.

closed 2023-01-03 by NikolajSchlej

selevo, 2023-01-03:
> Closing as a non-issue for UEFITool, but I can direct you to [this link](https://www.wikiwand.com/en/Serial_presence_detect#introduction), which explains how the SPD table should look like for a [...]

most of the data in the SPI BIOS is packed and cannot be found by direct comparison of the dump merged by the programmer. So the information on the link is completely useless and has nothing to do with the topic, all this I already know.

NikolajSchlej, 2023-01-03:
"Информация бесполезная, к теме отношения не имеет, я это все знаю уже." Информация как раз максимально полезная, к теме имеет отношение непосредственное, а вы, даже не попытавшись последовать вышеуказанной инструкции, начали мне рассказывать про то, что в прошивка упакована, и потому поиск байтовых последовательностей в ней не сработает, и вот это уже как раз к делу действительно отношения не имеет, т.к поиск в UEFITool работает по уже распакованным данным.

Вы, Сергей, приходите к разработчикам утилиты с вопросом, они вам отвечают, вы хамите в ответ. Понимаете, что если бы вы вместо хамства вы загрузили бы в эту тему дамп прошивки своего планшета, и сообщили модель оперативной памяти, которую он использует, у нас бы диалог начался с конструктива?