LongSoft/UEFITool #327: 7337f15ec9498d6e526f042dcb46a42220a47cf4 has broken ROM parsing

mikebeaton, 2023-01-02:
The most recent commit seems to have broken ROM parsing - this is a standard 144.0.0.0.0 Mac Pro ROM.

Before recent commit:

[image]

After:

[image]

NikolajSchlej, 2023-01-03:
Should be fixed in 19ab6076c48b2fc43856e784148fc571142f7891

closed 2023-01-03 by NikolajSchlej

mikebeaton, 2023-01-03:
Yes, fixed and appears to be displaying/extracting sections correctly again - tyvm