LongSoft/UEFITool #329: Identified some additional errors mistakenly introduced into guids.csv

mikebeaton, 2023-01-06:
I identified some further [g|G][A-Z] issues introduced in https://github.com/LongSoft/UEFITool/commit/0fb88ab76035df49fd7967f43e207b77c68ad363 , e.g. `SlingShot` -> `SlinShot`, `OemBadgingSupportDxe` -> `OemBadginSupportDxe`, also `BFGSmm` -> `BFSmm`.

Some have been fixed since, including by https://github.com/LongSoft/UEFITool/commit/1b143c89e6e8fd00af2dda2ab2455b68a3b2bfda .

Having identified the (a?) pattern for these, I have put together another PR fixing the (some?) remaining ones.

NikolajSchlej, 2023-01-08:
Looks like this is now fixed, closing as so.

closed 2023-01-08 by NikolajSchlej

mikebeaton, 2023-01-09:
Ah, I didn't check, I was expecting GitHub to automatically close it if the related PR was merged, since I mentioned the issue number in the PR comment. (Almost 100% certain I've seen that happen before, with the issue number where I put it, so not sure if GitHub have changed their settings, or if it's a per-repo setting.) Thx.