LongSoft/UEFITool #332: Does new engine support Cascade.

CrazyHulk, 2023-01-12:
The old engine not work for epc621d8a when I try to unlock 0xE2.
And it breaks modified images.

Does the new engine had fix this?

https://github.com/LongSoft/UEFITool/issues/91

NikolajSchlej, 2023-01-13:
Parsing is perfectly fine in new_engine, but builder is still not ready, so there's currently no way to perform modification with it.

Meanwhile, you can do the following:
- Open you file in UEFITool NE and extract BIOS region from it
- Use UEFIPatch to patch that BIOS region file
- Open the original file in a hex editor and replace the whole BIOS region in the original file with the result of UEFIPatch modification. 

In the file you've linked, BIOS region starts at exactly the middle (16Mb mark, offset 0x1000000).

closed 2023-01-13 by NikolajSchlej

crazyhulk, 2024-03-23:
Yes, new_engine parsing perfectly fine and the patch method works, but the 0xE2 will rewrite to 1 when wake from sleep.

In short, the CFG Lock will be enable after wake from sleep.

How did I found this?
1.  Open you file in UEFITool NE and extract BIOS region from it
2.  Use UEFIPatch to patch the BIOS region file
3.  Cut bios file  to keep bios start to offset 0x1000000 by dd command (as file A)
4.  Concat the file from step 3 and patched BIOS region file. (cat A B > C)
5.  Flash the BIOS
6.  Use RW Everything to check 0xE2, which be disabled as expected.
7.  Sleep and wake
8.  Use RW Everything to recheck 0xE2, which be enabled.

Is there something wrong?

NikolajSchlej, 2024-03-23:
If it gets re-enabled, there's a second piece of code (most likely in PEI or in S3 bootscript) that does that, and it also needs to be patched away. I'm currently traveling and don't have access to the tools required to investigate where that code is, so please remind me in a week about that.

crazyhulk, 2024-03-28:
Have a pleasant journey.

Have you finished your trip? 

@NikolajSchlej

NikolajSchlej, 2024-03-29:
Done now, will check on the weekend.

NikolajSchlej, 2024-04-04:
Checked that out, looks like the lock is getting set by CpuPei module during S3 wake. It is covered by BootGuard, so patching it out might make the board unbootable. Needs some testing, and likely can be patched a different way that doesn't trigger it. If you are willing to test (and have a SPI flasher to recover from broken BIOS), ping me here, I'll prepare a modded file.

crazyhulk, 2024-04-07:
I am willing to test, but I need another windows machine use SPI flasher to force flash bios. The test may cost more time.

crazyhulk, 2024-10-23:
> Checked that out, looks like the lock is getting set by CpuPei module during S3 wake. It is covered by BootGuard, so patching it out might make the board unbootable. Needs some testing, and likely [...]

Ping @NikolajSchlej .

Sorry, this matter has been put on hold for a long time. I can make  testing at any time now.
Looking forward to your mod file!

NikolajSchlej, 2024-10-24:
Put this on my todo list, will check on one of the next weekends.

crazyhulk, 2024-10-25:
Thx, waiting for your good news.​⬤

crazyhulk, 2025-01-07:
Happy new year! 🎉🎉🎉 @NikolajSchlej 

Has there been any progress?

NikolajSchlej, 2025-02-13:
None, completely forgot about this, life got busier lately. Can't promise anything other than keeping this in mind.

crazyhulk, 2025-02-14:
> None, completely forgot about this, life got busier lately. Can't promise anything other than keeping this in mind.

The busier life makes it easy to forget this.

Life first! Do not need promise anything!