LongSoft/UEFITool #334: Add support detect ACPI tables

slavonnet, 2023-01-18:
[LENOVO YOGA 2 PRO_VIUU3_NM-A074_Vina.zip](https://github.com/LongSoft/UEFITool/files/10440919/LENOVO.YOGA.2.PRO_VIUU3_NM-A074_Vina.zip)

in example BIOS 90% of "freeforn payload" is ACPI tables. You can add fast naming hack like "IntelIGopVbt" if payload have name of ACPI table (4 byte) and next bytes must equal "BodySize"

[image]

and list of names ACPI tables:

```
Known/Supported ACPI tables:
       1) AEST    Arm Error Source Table
       2) AGDI    Arm Generic Diagnostic Dump and Reset Device Interface Table
       3) ASF!    Alert Standard Format Table
       4) BDAT    BIOS Data ACPI Table
       5) BERT    Boot Error Record Table
       6) BGRT    Boot Graphics Resource Table
       7) BOOT    Simple Boot Flag Table
       8) CCEL    CC-Event Log Table
       9) CDAT    Coherent Device Attribute Table
      10) CEDT    CXL Early Discovery Table
      11) CPEP    Corrected Platform Error Polling Table
      12) CSRT    Core System Resource Table
      13) DBG2    Debug Port Table type 2
      14) DBGP    Debug Port Table
      15) DMAR    DMA Remapping Table
      16) DRTM    Dynamic Root of Trust for Measurement Table
      17) DSDT    Differentiated System Description Table (AML table)
      18) ECDT    Embedded Controller Boot Resources Table
      19) EINJ    Error Injection Table
      20) ERST    Error Record Serialization Table
      21) FACS    Firmware ACPI Control Structure
      22) FACP    Fixed ACPI Description Table (FADT)
      23) FPDT    Firmware Performance Data Table
      24) GTDT    Generic Timer Description Table
      25) HEST    Hardware Error Source Table
      26) HMAT    Heterogeneous Memory Attributes Table
      27) HPET    High Precision Event Timer Table
      28) IORT    IO Remapping Table
      29) IVRS    I/O Virtualization Reporting Structure
      30) LPIT    Low Power Idle Table
      31) APIC    Multiple APIC Description Table (MADT)
      32) MCFG    Memory Mapped Configuration Table
      33) MCHI    Management Controller Host Interface Table
      34) MPST    Memory Power State Table
      35) MSCT    Maximum System Characteristics Table
      36) MSDM    Microsoft Data Management Table
      37) NFIT    NVDIMM Firmware Interface Table
      38) NHLT    Non HD Audio Link Table
      39) PCCT    Platform Communications Channel Table
      40) PDTT    Platform Debug Trigger Table
      41) PHAT    Platform Health Assessment Table
      42) PMTT    Platform Memory Topology Table
      43) PPTT    Processor Properties Topology Table
      44) PRMT    Platform Runtime Mechanism Table
      45) RASF    RAS Features Table
      46) RGRT    Regulatory Graphics Resource Table
      47) RSDP    Root System Description Pointer
      48) RSDT    Root System Description Table
      49) S3PT    S3 Performance Table
      50) SBST    Smart Battery Specification Table
      51) SDEI    Software Delegated Exception Interface Table
      52) SDEV    Secure Devices Table
      53) SLIC    Software Licensing Description Table
      54) SLIT    System Locality Information Table
      55) SPCR    Serial Port Console Redirection Table
      56) SPMI    Server Platform Management Interface Table
      57) SRAT    System Resource Affinity Table
      58) SSDT    Secondary System Description Table (AML table)
      59) STAO    Status Override Table
      60) SVKL    Storage Volume Key Location Table
      61) TCPA    Trusted Computing Platform Alliance Table
      62) TDEL    TD-Event Log Table
      63) TPM2    Trusted Platform Module hardware interface Table
      64) UEFI    UEFI Boot Optimization Table
      65) VIOT    Virtual I/O Translation Table
      66) WAET    Windows ACPI Emulated Devices Table
      67) WDAT    Watchdog Action Table
      68) WDDT    Watchdog Description Table
      69) WDRT    Watchdog Resource Table
      70) WPBT    Windows Platform Binary Table
      71) WSMT    Windows SMM Security Mitigations Table
      72) XENV    Xen Environment Table
      73) XSDT    Extended System Description Table

Total 73 ACPI tables

```

NikolajSchlej, 2023-01-18:
You can already find the ACPI tables using Search and searching for the 4 bytes of the type listed above. Editing the tables requires Intel ACL or similar ASL/AML compilation/decompilation tool, so making UEFITool into an editor for them is too hard to bother. 

It could be possible to detect that a given raw file is an ACPI table by using some heuristics, but why exactly is it required, and what problem are we trying to solve by that, that can't be solved by Search? 

Можете отвечать по русски, если не хотите продираться сквозь английский. Непонятно, какую задачу решает эта предложенная эвристика, и какой именно вы видите "поддержку ACPI". Поддержка, это возможность делать что конкретно?

slavonnet, 2023-01-19:
Все просто. Продукты совершенствуются когда больше куски закрываются быстрыми кейсами. Сейчас в прошивке куча пустых мест которые при исследовании приходится каждое открывать в хекс редакторе и смотреть че там.  Когда биосов и десяток то запоминать в каких дырах что лежит вообще сложно. А тут можно взять готовый функционал пометки видео блока, запилить регу или маску и выводить юзеру типа Like ACPI Table.  Не надо править ,ASL/AML просто чуть помощи ) Все ACPI таблицы имеют фиксированный заголовк, сигнатуру из 4х символов таблицы, размер таблицы и ЦРЦ. Определяемость в блоках  будет 95%. Варианты смещений просто забить )

Количество всегда побеждает качество. Дыр много, можно потратить час-два и на 3-5% меньше пробелов будет. Такие продукты больше полезны как раз как смарт тулзы. Показать че где лежит, обзорность составить и погрузится во что-то. 

Немного лирики.... Было бы круто в конечном итоге раскладывать все на гит деревья, или VFS которое позволяло бы сравнивать че изменилось между биосами вместе с дизассемблером блоков. Как  пример - когда биос для сервера патчил, просто посмотрел модель старше  которая параллельно шла какое то время с моим . Составил карты "разницы" изменений в одних и тех же версиях биосоа. Выяснил специфичные блоки  для старого биоса и еще 2 года накатывал свежие фичи на не поддерживаемую систему.

NikolajSchlej, 2023-01-20:
Про сравнения: отлично помогает `File->Generate report...` на обоих файлах (или `UEFIExtract path/to/file report`), и сравнение полученных текстовых файлов diff-ом. Если нужно более качественное сравнение - `UEFIExtract` на оба и любая утилита для сравнения директорий с содержимым ('Beyond Compare', например).

Про количество и качество: это все отлично, но проблема в том, что если даже немного переборщить, то можно оказаться в ситуации `Nero Burning ROM`, и сдохнуть в атаке, поддерживая бесконечные фичи, 90% которых нужны буквально двум с половиной людям. Понятно, что определение ACPI-таблиц - это и несложно, и релевантно, и потому их проще сделать, чем спорить, особенно если не нужно парсить всю таблицу, а только общий заголовок.

Итог: убедил, сделаю, как руки дойдут.

timematcher, 2023-11-25:
This feature could be extremely useful to detect and extract ACPI tables..