LongSoft/UEFITool #335: Hex/Text search doesn't find in LZMA compressed section while GUID search works

xCuri0, 2023-01-27:
BIOS is from ASUS Z170-E version 3801

Searching ```PciBus``` in text search or ```CF8034BE``` (PciHostBridge allocation protocol) in hex search finds nothing. However searching ```CF8034BE``` in GUID search works and shows this.

[image]

There's a visible PciBus string so shouldn't atleast text search find it ?

NikolajSchlej, 2023-01-27:
You have the answer on your screenshot: "GUID pattern CF8034BE ... found as BE3480CF...".
The GUID of 00112233-4455-6677-8899-AABBCCDD is stored as "33221100554477668899AABBCCDD" because x86 is little-endian. It's not a bug in UEFITool, it's how UEFI GUIDs are stored on x86.

closed 2023-01-27 by NikolajSchlej

xCuri0, 2023-01-27:
@NikolajSchlej what about text search. It also finds nothing when I look for ```PciBus``` which is clearly visible. Usually it finds DXE modules by name but it doesn't find this one

NikolajSchlej, 2023-01-27:
Make sure to select Unicode for that UI sections text.

NikolajSchlej, 2023-01-27:
[image]

NikolajSchlej, 2023-01-27:
The name in your BIOS is not coming from an UI section, however, but from GUID database, that's why you can't find that text using "Search...".

NikolajSchlej, 2023-01-27:
[image]

NikolajSchlej, 2023-01-27:
You can unload the GUID database by using "File->Unload GUID Database", then (re)open your file and see that the names are now shown as GUIDs, and there's no "PciBus" unicode text to be found anywhere.

We do have some ideas on improving search and implementing GUID-database-assisted search, but it's a low priority feature currently.

xCuri0, 2023-01-27:
@NikolajSchlej thanks for explaining why the text search won't work