LongSoft/UEFITool #336: Seg fault with qt6 + new_engine

4e4o, 2023-01-30:
With qt6 i have segfault [here](https://github.com/LongSoft/UEFITool/blob/new_engine/common/nvramparser.cpp#L1509)
According to c++ strict aliasing rules type casting from byte array to char16_t is UB. In most cases it Ok, but as you can see not always. In x86 case you must ensure that char16_t* points to address aligned to 2. Here is my opened issue on qt bug tracker https://bugreports.qt.io/browse/QTBUG-110741

[image]

NikolajSchlej, 2023-01-30:
Nice find, C++ remains to be fun. The underlying data in unaligned by design, so we'll have to work around that by making copies or use different API calls.

NikolajSchlej, 2023-01-30:
Also, I bet there's a metric shitton of other UB cases, makes sense to start using UBSAN to catch them.

NikolajSchlej, 2023-01-30:
Could you share the exact environment, compiler version and input file, so I could reproduce the crash locally?

4e4o, 2023-01-31:
> Could you share the exact environment, compiler version and input file, so I could reproduce the crash locally?

Sure. 

fedora 37 x86_64
uname -a
Linux mobile 6.1.7-200.fc37.x86_64 #1 SMP PREEMPT_DYNAMIC Wed Jan 18 17:11:49 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux
checked clang and gcc, same behaviour
clang version 15.0.7 (Fedora 15.0.7-1.fc37)
gcc version 12.2.1 20221121 (Red Hat 12.2.1-4) (GCC)

You can simply run minimal example from [here](https://bugreports.qt.io/browse/QTBUG-110741). Also fw.bin attached.  
[01966.bin.zip](https://github.com/LongSoft/UEFITool/files/10543473/01966.bin.zip)

NikolajSchlej, 2023-02-01:
Should be fixed by https://github.com/LongSoft/UEFITool/commit/c9939e23ec7c758166ed020f5e410a67e846dc0c

closed 2023-02-01 by NikolajSchlej