LongSoft/UEFITool #338: Segfault while loading firmware for Lenovo Yoga 7

stuarthayhurst, 2023-02-03:
I've extracted the UEFI firmware, with innoextract and InsydeImageExtractor, but this tool gives a segfault while trying to load it.

I've uploaded the file for troubleshooting:
[out.zip](https://github.com/LongSoft/UEFITool/files/10583449/out.zip)

NikolajSchlej, 2023-02-04:
[image]
Did not segfaulted for me. Could you share the more information on the OS and tool versions, so we can reproduce it?

stuarthayhurst, 2023-02-04:
Hm, I'm on an up-to-date Debian Sid (`uefitool` version `0.28.0+A62-1`), but it also happened with the current flatpak version

NikolajSchlej, 2023-02-04:
Confired the crash of the latest flatpak build on Fedora 37, investigating.

NikolajSchlej, 2023-02-04:
Same crash as #336, should be fixed by the ToT commit. You cam download the artifact from Github Actions for this repo or wait for a new release.

> Thread 1 "uefitool" received signal SIGSEGV, Segmentation fault.
0x00007ffff6848c99 in QtPrivate::qustrlen(char16_t const*) () from /usr/lib/x86_64-linux-gnu/libQt6Core.so.6
(gdb) bt
#0  0x00007ffff6848c99 in QtPrivate::qustrlen(char16_t const*) () at /usr/lib/x86_64-linux-gnu/libQt6Core.so.6
#1  0x00007ffff68588e8 in QString::fromUtf16(char16_t const*, long long) () at /usr/lib/x86_64-linux-gnu/libQt6Core.so.6
#2  0x0000555555591aa9 in NvramParser::parseVssStoreBody(QModelIndex const&, unsigned char) ()

NikolajSchlej, 2023-02-04:
This is the link to that artifact, need to be logged in to GitHub to download it: https://github.com/LongSoft/UEFITool/suites/10703275646/artifacts/536038575

closed 2023-02-04 by NikolajSchlej