LongSoft/UEFITool #341: Search doesn't work in UEFITool A64 Mac?

nightgolfer, 2023-02-15:
Downloaded [UEFITool_NE_A64_universal_mac.zip](https://github.com/LongSoft/UEFITool/releases/download/A64/UEFITool_NE_A64_universal_mac.zip) last night and did a search for "CFG Lock" (text) in downloaded BIOS firmware, no results.

Downloaded an earlier version of firmware that I know has "CFG Lock" in it, same results, nothing found.

Downloaded UEFITool A63, search in both firmwares yields expected search results.

Monterey 13.6 (OpenCorer 0.8.8).

The two firmwares in question: 
[Gigabyte Z270X UG - F9d](https://download.gigabyte.com/FileList/BIOS/mb_bios_ga-z270x-ultra-gaming_f9d.zip?v=e44601ffb8e754675485a499b86047ae)
[Gigabyte Z270X UG - F7](https://download.gigabyte.com/FileList/BIOS/mb_bios_ga-z270x-ultra-gaming_f7.zip?v=cb062b95421a16aff637f6ae49e4d80a)

One thing I noticed — A64 Mac is tagged as "universal", A63 is not. Not sure if this is relevant.

NikolajSchlej, 2023-02-15:
This is likely caused by our switch to Qt6, will investigate.

NikolajSchlej, 2023-02-15:
More specifically, ASCII text search works, but Unicode does not.

NikolajSchlej, 2023-02-16:
Also confirmed on Linux, Unicode search doesn't work with Qt6 across the board, most likely due to changes in Regex engine.

NikolajSchlej, 2023-02-17:
Should be fixed by https://github.com/LongSoft/UEFITool/commit/6f9dc0ab885a34330ef361c61226994be44943c4

closed 2023-02-17 by NikolajSchlej