LongSoft/UEFITool #343: Add help or version parameters at UEFIExtract and UEFIFind with return code 0

platomav, 2023-02-15:
It is currently not possible to quickly verify the successful installation of the UEFIExtract and UEFIFind executables because they offer no "help" or "version" parameters.

Calling the executables without any parameter, does show the help screen, but the return code is 1 instead of 0.

The only workaround, I have found, is to use the following instead:

``` bash
(uefiextract || [ $? -eq 1 ])
```

NikolajSchlej, 2023-02-15:
Will add both `-v/--version` and `-h/--help`, this is a legacy of Windows-only days.

NikolajSchlej, 2023-02-16:
Fixed in https://github.com/LongSoft/UEFITool/commit/cb6ef45d0ca2134ae517ca017c89f0fe23a861a6

closed 2023-02-16 by NikolajSchlej

platomav, 2023-02-16:
Tested all variations (-v, --version, -h, --help) and everything is good now, including the exit code of 0. Thank you!