LongSoft/UEFITool #346: AMI Aptio 2.17.1255 header parser bug in A64

StackerDEV, 2023-02-21:
I came across this playing with old junk..

**ROM Bin dumped:**
https://mega.nz/file/JYZkCKyY#4wGqzIvyKimRUhpG2vsB-2CCrirfdaasylJrvxpNRZg

**Problem description**:
Extract as is and replace body is working good in UEFITool 0.28.0 for above rom. I did the mod, eeprom and system posting/booting properly no issues.
When I tried in A64 header bugged (corrupted), can't replace anymore I flash rom sys no post. I have no H80 display connected yet on backorder, trying to make a SFF system from junk mb. Any other options I have not messed with using AMICBP for improvering thermal tripoints and unlocking more options for full life extension.. All I did search BMP magic change bootlogo..... Should be super easy effortlessly but somehow not possible with this rom in the new version A64, try ir yourself.

Details of what I am trying to do.
->>

**Machine**
Rom name: Terra-Clevo-Medion_1530_i3-6100U.rom
PCB revision: MA50LP_v1.0 SKL PCH-LP Mobile(U) 21/C1 Premium SKU PCH RC 1.9.0.0.
AMI Aptio ver: 2.17.1255
Aptio core ver: 

### **Info**:

File GUID: 4B1E6486-6C0E-538E-1357-8FB97063DFD6
Type: 02h
Attributes: 00h
Full size: 23BFh (9151)
Header size: 18h (24)
Body size: 23A7h (9127)
State: F8h
Header checksum: 34h
Data checksum: AAh

### RAW SECTION
Extract as is ..

**header + bmp data**
D2 E7 3D 19 & 42 4D CE ********

### HEADER
**FILE MAGIC data after RAW section dump:**
42 4D CE
BM
CS5 header no good, magic must be 42 4D CE not 42 4D B8 or no post.

NikolajSchlej, 2023-02-21:
It's very unclear from this report what the issue is, sorry.
I did found a bug in that "Extract body..." doesn't work for some reason and that will be fixed in A65 for sure, but I need a better explanation to understand what's up.

NikolajSchlej, 2023-02-21:
The fact that A64 can not replace is known, as the editing part is still not done in the new engine, tracked by #67.

NikolajSchlej, 2023-02-21:
The section itself looks fine and has the expect BPM image
[image]
 magic

NikolajSchlej, 2023-02-22:
I've extracted the body of that raw section in 0.29.0 and a fixed version of A65 (https://github.com/LongSoft/UEFITool/commit/62d96a16df028fa913f5f9a2ce4b43ba743c5f6a), and the files are identical.
```
% shasum -b 0280.raw 
f37d66c72b3ce3664541771adfe36b817ce547b1 *0280.raw
% shasum -b A65.raw 
f37d66c72b3ce3664541771adfe36b817ce547b1 *A65.raw
```

NikolajSchlej, 2023-02-22:
Closing this as "can not reproduce", please reopen it if you do not agree.

closed 2023-02-22 by NikolajSchlej

StackerDEV, 2023-02-22:
That seems odd (I remember at Alpha 64 there was a odd situation header was corrupted when I write back and look again with UEFITool it was that way around). Next time I try to make it reproduceable, but I was aware of possible error still. Project needs more reversing and that takes time and lots of testing.

I might have made an error at some point in the process.
Thanks for investigating and then amazing project. Will try to support it if possible.