LongSoft/UEFITool #350: I would like to see the Replacebody feature enabled for UEFITOOL NE

Ejkxojel, 2023-03-06:
I am Japanese. Although it is a translated sentence, there is a problem with **UEFITOOL**.
Lenovo G500 BIOS update ROM file
"Winviwgp203.rom" was used mainly for wifi whitelist disabled. I wanted to use Replace Body because I was able to rewrite, but it was grayed out and I could not do it.
With 0.28.0, you can't even read the ROM file, and you get an error "Invalid Uefi Volume".
There is no error in **NE**, but Replace Body cannot be used.
Please be able to use **Replace Body**

NikolajSchlej, 2023-03-06:
#67

closed 2023-03-06 by NikolajSchlej

NikolajSchlej, 2023-03-06:
I'd also like to suggest using [PhoenixTool](https://forums.mydigitallife.net/threads/tool-to-insert-replace-slic-in-phoenix-insyde-dell-efi-bioses.13194/) for cases where modification is required but UEFITool 0.28.0 doesn't work. You need to register on the forum to see download links.