LongSoft/UEFITool #352: UEFITool new engine cannot find a volume found in the old engine

mytbk, 2023-03-11:
firmware link: https://ftp.hp.com/pub/softpaq/sp102001-102500/sp102456.exe
SHA256: f9e8c0e7083e0ca745bfc6762ae2ac9c031ea6793cd009174b1dec2358c699cf
Use ``unar sp102456.exe`` to get the UEFI image sp102456/M71_0131.bin.

When using the UEFITool old engine, expand the first FFSv2 volume, under which is a file with GUID 4A538818-5AE0-4EB2-B2EB-488B23657022 (subtype DXE core), then expand all the things under it, can get another FFSv2 volume with GUID 7A9354D9-0468-444A-81CE-0BF617D890DF, which contains all the DXE modules of the firmware. When using the new engine, after expanding the compressed section and the raw section under it, cannot get the FFSv2 volume, instead the new engine shows a padding with a different size as in the old engine, and an NVRAM volume following it.

NikolajSchlej, 2023-03-12:
Can not reproduce, the volume is there for me in NE A65.

[image]

mytbk, 2023-03-12:
I did a bisect between A65 and git new_engine. b8567d32cc158eb68d900d9a161e92889e643627 is the first bad commit.

vit9696, 2023-03-12:
Which operating system do you use? Which binary file?

mytbk, 2023-03-12:
@vit9696 I'm using Arch Linux x86_64. The binary file is the one mentioned in this issue.

vit9696, 2023-03-12:
UEFITool binary file, do you self-compile?

NikolajSchlej, 2023-03-12:
Confirmed, the fix is to remove the offending commit and won't attempt to fix any further issues without a test suite, or stuff like this will regress again.

NikolajSchlej, 2023-03-12:
Fixed, will remove that pair from the history later, thanks for catching this before it got into A66.

closed 2023-03-12 by NikolajSchlej