LongSoft/UEFITool #354: failure parsing dell R440 .cap file

greyltc, 2023-03-26:
Hi. Thanks for the really cool tool!

I've tried to use the latest release here https://github.com/LongSoft/UEFITool/releases/download/A65/UEFITool_NE_A65_win64.zip on a .cap firmware file (R440-021701C.cap) provided by Dell for my system here https://www.dell.com/support/home/en-us/drivers/driversdetails?driverid=5kdwv

But when I try to load up the .cap file, I get a parser error: `FfsParser::parse: not a single Volume Top File is found, the image may be corrupted`

Am I doing something wrong or is there a workaround for this?

NikolajSchlej, 2023-03-26:
That file is not an UEFI PI specification compatible Capsule, but (yet another) Dell invention. It has a copyright header of 0xA8 bytes, followed by LZMA-compressed container consisting of 2 full BIOS images, and several other smaller BLOBs. The images can be extracted with a hex editor, and then parse with UEFITool NE just fine. 
[R440-021701C.bin.zip](https://github.com/LongSoft/UEFITool/files/11072290/R440-021701C.bin.zip)

@platomav, looks like we need a "Dell CAP Extractor" script added to your https://github.com/platomav/BIOSUtilities collection.

closed 2023-03-26 by NikolajSchlej

greyltc, 2023-03-26:
That's great! Thank you for teaching me. I don't understand why you closed this issue though. Are you saying that functionality to load in .cap files from Dell is out of scope for this tool?

NikolajSchlej, 2023-03-26:
It is indeed out of scope of this tool to support vendor-specific wrappers around UEFI PI-compatible images.

> **Known issues**
>Some vendor-specific firmware update files can be opened incorrectly or can't be opened at all. This includes encrypted HP update files, Dell HDR and EXE files, some InsydeFlash FD files and so on. Enabling support for such files will require massive amount of reverse-engineering which is almost pointless because the updated image can be obtained from BIOS chip where it's already decrypted and unpacked.

greyltc, 2023-03-26:
It's very confusing that the tool is designed to support some .cap files but not all of them. If the tool throws an error reading some random .cap file I give it, how should I know if that's a bug or if that's expected (/intended?) behavior?

If handling many different types of vendor bios archives is very difficult, maybe the tool could gain a plugin architecture that allows 3rd parties to add support for loading many different types of archives without expanding the scope of this project too much?

NikolajSchlej, 2023-03-27:
There's a spec-compatible Capsule format called UEFI Capsule, and it uses .cap extension. This format is well-defined, mostly vender-agnostic, and therefore, supported. Vendor-specific stuff is not.

Extraction scripts like BIOSUtilities linked above cover the "extract something sane from whatever vendor-specific container" part, and that's enough. Plugins is a nice idea, but the project is already OSS, if you want to extend it - a fork is one button press away, pull requests are always welcome.

NikolajSchlej, 2023-03-27:
To answer your original question:
>If the tool throws an error reading some random .cap file I give it, how should I know if that's a bug or if that's expected (/intended?) behavior?`
Report the error and we will investigate together. You did that already in this issue, it worked well enough.

platomav, 2023-03-29:
> @platomav, looks like we need a "Dell CAP Extractor" script added to your https://github.com/platomav/BIOSUtilities collection.

Yeah, I've been doing that exact process manually for years now with those .cap images. Some of them are encapsulated in an UEFI volume/capsule (don't remember), but most of them are bare and start with the Remote BIOS Update (RBU) header, which is often 0x54 in size and the LZMA streams follows afterward. The decompressed output is indeed some sort of multi-payload arrange (isn't Dell tired from all the countless proprietary formats they have developed already? 🙄).