LongSoft/UEFITool #355: I need to change BIOS Settings from Command Line (QProcess)

SlimRG, 2023-03-26:
I have different uefi bios images.
I need to change Optimal values:
In: _"Common RefCode Configuration > IntelRCSetup > Advanced Power Management Configuration > CPU C State Control"._
"Package C State limit" - "C2 state"
"CPU C3 report" - "Enable"
"CPU C6 report" - "Disable"

I can do it by AMIBCP5, but it has only GUI with bad UI for SPY++;
So I can't automate usage of this app.

After that I found your app.
But don't know - how to do, what I want.

SlimRG, 2023-03-26:
Instruction and roms you can find here:
https://github.com/Koshak1013/HuananzhiX99_BIOS_mods

NikolajSchlej, 2023-03-26:
I would not recommend to use UEFITool for this task because currently we do not support a version that can perform modifications (0.2x) and the one we do support lacks that capability. Try using PhoenixTool instead.

closed 2023-03-26 by NikolajSchlej