LongSoft/UEFITool #356: Improve Intel Microcode header parsing

NikolajSchlej, 2023-03-27:
Intel recently updated their microcode header format to support In-Field Scan, and MCExtractor already got support for it here:
https://github.com/platomav/MCExtractor/commit/32de67472e3092505ce290f573cb587c06d48344 as a fix for https://github.com/platomav/MCExtractor/issues/60

Some changes need to be made to improve detection of new microcodes, and it's likely time to rewrite it's parser into KaitaiStruct.

NikolajSchlej, 2023-03-27:
CC @platomav, who likely has samples of the new files, and maybe even BIOS images with them.

platomav, 2023-03-29:
Here is a sample UEFI with microcodes which utilize the Minimum Version field of the microcode:

[BIOS_X13SEW-1C51_20230119_1.1_STDsp.zip](https://github.com/LongSoft/UEFITool/files/11105875/BIOS_X13SEW-1C51_20230119_1.1_STDsp.zip)

For the Microcode Header Type 2 (i.e. IFS), according to the [kernel docs](https://docs.kernel.org/x86/ifs.html), Intel is supposed to publish samples at their GitHub, but the link is not currently valid/present (https://github.com/intel/TBD).

NikolajSchlej, 2023-03-30:
By TBD they mean To Be Determined, i.e. they will publish it, but right now the true URL is unknown. Hope they update that documentation with a correct link later.
IMO, reuse of microcode header for IFS firmware is bad idea, because the header famously lacks a magic value and requires a ton of heuristics to both detect it in raw input, and to not detect something else as it in raw input. I understand that having FIT point at the precise location does help the CPU to load it, but for heavens sake, Intel, PLEASE USE A BETTER MAGIC THAN 0x1 and 0x2!

platomav, 2023-03-30:
From what I read, IFS had a v1 which was using a format very similar to microcode but not the same. Which was worse because you have a slight copy of the same thing but modified in ways which will break general parsers. Same with kernel drivers logic. With v2, they integrated both into the official microcode structure by changing Header Version (1) to Header Type (1 mc, 2 ifs) and introducing the Metadata length and equivalent area in the microcode blob:

[image]

Generally, Intel tends to be better than others when it comes to adding magic values, but not perfect (Microcode, IFWI/BPDT, CSE Layout Table come to mind). From my (bad) experiences at MCE, the worst is AMD... 😨

closed 2025-04-14 by NikolajSchlej

platomav, 2025-04-14:
Good call, I had removed my (guess-work) code for IFS because it was causing crashes without an actual sample to test on.