LongSoft/UEFITool #361: can you make a uefiextract for uefi oprom? for linux? for example

kocoman1, 2023-05-06:
can you make a uefiextract for uefi oprom? for linux? for example 
https://www.insanelymac.com/forum/topic/299627-uefi-option-rom-extractor-and-decompressor-v10/

i using it for the sas oracle/nynt/seagate f40/f80 warpdrive uefi oprom extraction for booting off its 100gb-128gb etc 
but my windows install and osx install is always broken i prefer linux if possible.

thx

NikolajSchlej, 2023-05-07:
That utility can be ported to pure C fairly easily (as it was mostly ported from C by copy-pasting). It's not an issue of UEFITool though, so I will close this as N/A. Will ping you here when that port is ready.

closed 2023-05-07 by NikolajSchlej

NikolajSchlej, 2023-05-08:
@kocoman1, here, a single C file that should be buildable on any OS with a C compiler and libc (Linux, macOS, Windows, whatever): [UEFIRomExtract.zip](https://github.com/LongSoft/UEFITool/files/11417251/UEFIRomExtract.zip)
Usage: download, unzip, run `gcc UEFIRomExtract.c -o uefiromextract`, replace `gcc` with `clang` if your Linux distro uses it.
Tested locally on Ubuntu 23.03, worked for me (with minor compiler warnings that can be ignored).

b23prodtm, 2023-08-20:
Need it to compile for macOS, it run fine on monterey 12. https://forums.macrumors.com/threads/pre-opencore-gop-support-for-efi-era-imacs-and-mac-pros.2378942/

NikolajSchlej, 2023-08-21:
Install Xcode and compile it using the same command as described above.