LongSoft/UEFITool #362: Intel board MGM45WU extraction issues

swiftgeek, 2023-05-07:
I don't really get it, but I feel like some files are broken, while others extract correctly.
But I think support for those firmware files greatly improved in last few years, so I felt like maybe mentioning it would lead to some new findings.

[MGM45WU.zip](https://github.com/LongSoft/UEFITool/files/11413745/MGM45WU.zip)

NikolajSchlej, 2023-05-07:
The file is of 2008 vintage, when things had rough edges and UEFI was still missing the Unified part. Some things did change since then, but this Framework-compatible file is a piece of legacy now, and it's unclear what value will adding proper support to all of its quirks actually brings in 2023. Thanks for reporting though.

closed 2023-05-07 by NikolajSchlej

swiftgeek, 2023-05-07:
To me the value is that it's split it up better after extraction than legacy bios (from those that did extract correctly) and it is easier to read in ghidra decompiler (not that I'm any good at doing things in ghidra). I'm also still using that platform of that generation extensively, but couldn't find any other instance of (U)EFI for that intel generation (or even several next ones) with all the vPro features, built for (U)EFI rather than running some of them under CSM (like what happens with HP Elitebook 2530p, which also seems to extract way better now than in 2019). 

If it isn't too much of a bother, any pointers as to what is going wrong with that file / what further resources could be read about pre-UEFI quirks/specs? I feel like the most broken things are with freeform subtypes, but I only make that wild guess due to how they end up being so big after extraction

NikolajSchlej, 2023-05-08:
The image is mostly conforming to early Framework specs listed here: https://www.intel.com/content/www/us/en/architecture-and-technology/unified-extensible-firmware-interface/efi-specifications-general-technology.html
There are several quirks that are Intel-specific and don't really worth bothering about because .BIO files aren't really UEFI images, they are update capsules. The real image can be dumped from SPI flash via Intel Flash Programming Tool, flashrom or hardware SPI programmer, then opened in UEFITool. I expect most parsing issues to be gone, and the rest will be something like "Intel decided to take a section type that was unused and use it for their own thing, then UEFI spec defined it as something else entirely and UEFITool gets confused".