LongSoft/UEFITool #363: UEFIExtract fails to recursively extract volumes within Non-UEFI section

platomav, 2023-05-10:
I encountered some slightly odd UEFI capsules which do include valid UEFI volumes but within areas which are shown as Non-UEFI:

[image]

UEFITool is able to detect/parse valid UEFI volumes within Non-UEFI data. However, UEFIExtract is having trouble extracting them. The old-style "unpack" command does work, but not "dump" and, by extension, "all":

[image]

I am attaching 14 sample capsule images, which should help to narrow down this UEFIExtract behavior.

[Non_UEFI.zip](https://github.com/LongSoft/UEFITool/files/11446361/Non_UEFI.zip)

NikolajSchlej, 2023-05-11:
This looks like an issue with duplicated GUIDs and writing stuff into the sam folder over and over and over again, just from the error messages. "unpack" working also indicates that.

platomav, 2023-05-11:
Ha, interesting, did not notice the vintage factor. And here I thought one of the oldest public implementation was Apple's iMac from early 2006 (attached for reference). Thanks for looking into it whenever time permits, Nikolaj!

[IM41_88Z_0039_B00_2006-01-06_12-06_B483979D.zip](https://github.com/LongSoft/UEFITool/files/11456481/IM41_88Z_0039_B00_2006-01-06_12-06_B483979D.zip)

platomav, 2023-05-17:
I'm also uploading another much newer sample, which has volumes withing "padding":

[image]

Manually extracting the padding area and feeding it to UEFITool, reveals the following structure:

[image]

[233baa0298ff7e9d46479868bbeb45948cbf649f_BIOS.zip](https://github.com/LongSoft/UEFITool/files/11502439/233baa0298ff7e9d46479868bbeb45948cbf649f_BIOS.zip)

NikolajSchlej, 2023-05-18:
The later one is a combination of several unexpected interactions between NVRAM parser and FFS parser, and will likely be fully resolved after either both or just the NVRAM one fully moves to KaitaiStruct. Thanks for catching it though.

platomav, 2023-05-18:
Great, thanks for the check. Happy to help. If you need me to create a separate issue for the "BIOS.cap" sample from yesterday, I can do so.

Also, here is the link to the equivalent ticket I opened for Intel at Chipsec, which has a lot of interesting discussions and tests/results on those samples, which could be useful when the fixes on UEFITool are scheduled in the future:

https://github.com/chipsec/chipsec/issues/1790

NikolajSchlej, 2023-06-07:
Turns out the main reason for that "Error 35" issue is the dreaded `MAX_PATH` limitation for Windows that needs to be fixed by reimplementing `_stat`, `_mkdir`, `_rmdir`, `_chdir` and `_fullpath` using `CreateDirectoryW` and similar functions with expanded WCHAR paths prefixed by `\?\\`.

I've tried to find a suitable implementation, and Ruby project [seems to have one](https://github.com/ruby/ruby/blob/master/include/ruby/win32.h#L168) that is BSD 2-cl licensed, so we can borrow it.

NikolajSchlej, 2023-06-20:
Should be partially fixed by https://github.com/LongSoft/UEFITool/commit/13f38aa4d71ccc1fd3335ac1e190eddfeeb85bdd
However, for the fix to start working, you need Windows 10 1067+ or Windows 11, and add the following to your registry:
```
Windows Registry Editor Version 5.00

[HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Control\FileSystem]
"LongPathsEnabled"=dword:00000001
```
More information on the Windows-side of the fix here: https://learn.microsoft.com/en-us/windows/win32/fileio/maximum-file-path-limitation?tabs=registry

closed 2023-06-20 by NikolajSchlej

NikolajSchlej, 2023-06-20:
[image]

platomav, 2023-06-20:
I can confirm that it works now. Both Windows Explorer and Python work as expected. Long paths were enabled before, but it needed support from the program-side. Thanks Nikolaj!

These long paths work in "funny" ways. Once they get triggered, Windows Explorer context menu breaks a bit, and you cannot right-click to open files or even launch the terminal. Nothing happens. However, you can modify the path at the top and remove the long path UNC prefix (i.e. "\\?\"), which causes it to auto-convert the path to the old-style DOS short names (with the tilde, ~) and then all normal context menu actions start working again. So weird. This is obviously not an UEFITool behavior but Windows stupidity. No one really explains these side effects anywhere, but I did find one person who saw the same weirdness [here](https://answers.microsoft.com/en-us/windows/forum/all/lesser-known-effect-of-path-length-limitation-in/383b8f71-c425-4869-8d5f-654d56899dc4).

Anyway, something else to note: On Windows 10+, the MAX_PATH can indeed be increased but the limit for MAX_NAME (or whatever it is called) is still ~250 characters. So even if the whole path can now be extended, the final path component (i.e. file/link/folder name) cannot exceed that (small, by today's standards) hard-coded value.

NikolajSchlej, 2023-06-20:
Those UNC (not actually, more like "transparent" as in "sent to device driver as is without processing") paths are the worst. I've tried to fix this bug without requiring registry fixups and Windows 10+, but bailed really hard because reimplementing all the path normalization just for Windows is too much, especially given that we do already have a special extraction mode that lays files in a flat structure anyway, so that problem is a bit less glaring. 
Still, that limitation was and remains stupid, and I'm glad I haven't yet stumble upon anything like this in Linux or macOS.

platomav, 2023-06-21:
> especially given that we do already have a special extraction mode that lays files in a flat structure anyway, so that problem is a bit less glaring

You mentioned that the duplicates' problem was also fixed, but I forgot to check. Can indeed confirm that both "unpack" and "dump" produce the same files with a different structure. I see that an index is added to duplicates (001, 002 etc). Technically, though, the "all" mode does store some additional data (the raw body/header), which are not available in the flat (i.e. "unpack") mode. Not a big deal, but just something I noticed.

> I'm glad I haven't yet stumble upon anything like this in Linux or macOS.

I actually managed to hit that limit on Linux (Ubuntu) a few weeks back (not with UEFITool). Turns out that technically there is no FS limit, but most Linux distros do impose the MAX_PATH constant, which happens to allow 4096 characters. Supposedly, it can be increased with a command but in my case it would still not allow paths longer than 4096 characters so who knows, haha. Still, much better than the crapy/old Windows default or semi-working extended behavior. And if I manage to hit the limit with UEFITool, I'll let you know. 😉