LongSoft/UEFITool #366: Need to search for FFS volumes inside non-AMI NVRAM areas

NikolajSchlej, 2023-06-20:
@platomav recently found this nasty file that happens to hide `LenovoComputraceLoader`, `LenovoComputraceEnabler` and several other DXE drivers inside a not-so-well-formatted NVRAM area. Need to support this corner case.
[FFS_volume_in_NVRAM_area_BIOS.zip](https://github.com/LongSoft/UEFITool/files/11794988/FFS_volume_in_NVRAM_area_BIOS.zip)

NikolajSchlej, 2025-03-07:
Fixed by 2e01e315168002ee5f0c33d3088fc88a63ce442b, will be delivered in A70 together with all the other NVRAM parsing improvements.

closed 2025-03-07 by NikolajSchlej

NikolajSchlej, 2025-03-13:
Released as part of A70.

platomav, 2025-03-13:
Beautiful, great work!!

[image]