LongSoft/UEFITool #371: Compressed sections unsupported

szymor, 2023-07-18:
It seems I cannot uncompress compressed sections. The example:

```
Fixed: No
Base: 185060h
Offset: 18h
Type: 01h
Full size: 343Ch (13372)
Header size: 9h (9)
Body size: 3433h (13363)
Compression type: 01h
Decompressed size: 4F94h (20372)
Compression algorithm: Tiano
```

Is there a way to extract the contents of this section outside of UEFITool?

szymor, 2023-07-18:
It seems UEFITool is able to uncompress the section. I just expected different contents. I will wait for the reply, but I think the ticket can be closed.

NikolajSchlej, 2023-07-18:
Unclear what you mean by "can't uncompress" here, because it would not be detecting the algorithm as "Tiano", if decompression would fail for it.

closed 2023-07-18 by NikolajSchlej