LongSoft/UEFITool #376: Off-by-one error preventing parsing of Boot5 partition in IFWI 1.6 and 1.7 regions

NikolajSchlej, 2023-08-25:
Finally found an image in the wild that has Boot5 region populated (and using legacy $FPT format).

```
    // Add boot partitions
-   for (UINT8 i = 0 ; i < 4; i++) {
+   for (UINT8 i = 0 ; i < 5; i++) {
        if (ifwiHeader->BootPartition[i].Offset != 0 && ifwiHeader->BootPartition[i].Offset != 0xFFFFFFFF) {
            IFWI_PARTITION_INFO partition = {};
            partition.type = Types::IfwiPartition;
```

XutaxKamay, 2023-09-04:
> Finally found an image in the wild that has Boot5 region populated (and using legacy $FPT format).
>
> ```
>     // Add boot partitions
> -   for (UINT8 i = 0 ; i < 4; i++) {
> +   for (UINT8 i = 0 ; i < 5; [...]

Can you share the image in question?

NikolajSchlej, 2023-09-08:
[MAIN.zip](https://github.com/LongSoft/UEFITool/files/12564065/MAIN.zip)
[image]

commit 6f9a4c0d (2023-11-02): Fix off-by-one error in parsing IFWI partition table

Fixes #376
closed 2023-11-02 by NikolajSchlej