LongSoft/UEFITool #377: Extract bitmap image

josephernest, 2023-09-18:
Congrats @NikolajSchlej and others for creating UEFITool!

Would you know how to locate the "Phoenix bitmap image logo" (maybe a BMP or TIFF or JPG) in a UEFI BIOS?

I would like to replace it by another logo.

I don't find it here: 

[KTQ8787.zip](https://github.com/LongSoft/UEFITool/files/12647743/KTQ8787.zip)

Would you know where it is?

Thanks in advance!

NikolajSchlej, 2023-09-18:
The logo is in GIF89a format, stored in a freeform file with GUID 418E3259-AE5A-4F09-8936-74C78BA27D6C inside a Raw section. The section also has a 0x29 bytes header with several image size entries that need to be adjusted if the new logo has a different size. 

There's also allegedly a tool that can automate logo replacement for such BIOS images, found it by searching for the GUID above: https://www.fontana.com.ar/2018/04/14/splashhack-v-0-1-beta/ 
Haven't tested it, so can't say how well it works.

This is the default logo extracted from the image above:
[image]

NikolajSchlej, 2023-09-18:
To answer the obvious next "how did you find it" question, most image formats have magic bytes that help parsers decide how to parse a given image. In this case, the magic is "GIF89a", which can be found by using File->Search->Text, and unsetting the default Unicode checkbox. Such text is then found in 3 files: the one mentioned above, a raw section inside a driver called SystemOsb (114BC6C1-404B-4232-8316-7CFA66D0B60A), and PE32 executable section inside a driver called SystemImageDecoderDxe (5F65D21A-8867-45D3-A41A-526F9FE2C598). Obviously, the last one is the decoder, so the image we are searching is not there. 
Similarly, JPEG images can be found by ASCII magic "JFIF", BMP images by "BM" and so on.

josephernest, 2023-09-18:
Thanks a lot @NikolajSchlej!

Good idea for `GIF89a`, I didn't tried this :)
I tried with:
* HEX `4D 4D 00 2A` and `49 49 2A 00` for TIFF files, without success
* `BM` for BMP files, but there were too many results

You're right: 

418E3259-AE5A-4F09-8936-74C78BA27D6C is the Phoenix logo
114BC6C1-404B-4232-8316-7CFA66D0B60A is the Intel logo  (I extracted it)
5F65D21A-8867-45D3-A41A-526F9FE2C598 is probably the decoder executable

About  [SplashHack](https://www.fontana.com.ar/2018/04/14/splashhack-v-0-1-beta/), I am not sure it will work, because this seems to require 2 images (one "splash" and one "logo"; in my case there are no 2 images, but just one (in my case the Intel logo is not related)). Also this tool uses 2 JPG files, while here we need GIF files.

Do you think it would be possible to just replace the Raw body of `418E3259-AE5A-4F09-8936-74C78BA27D6C` with UEFITool with a new GIF?

NikolajSchlej, 2023-09-18:
Yes, it's possible to just replace that body and adjust some sizes in the header, but there are many images inside 418E3259-AE5A-4F09-8936-74C78BA27D6C, so you need do that replacement carefully, so that replacing one logo in the beginning won't also remove all the other images you had no plans to remove. Also, UEFITool NE doesn't support image editing (yet), so please use either UEFITool 0.28.0, or latest PhoenixTool for it.

josephernest, 2023-09-18:
Thanks for your answer!  You're right, there are 6 GIF images at this GUID 418E3259-AE5A-4F09-8936-74C78BA27D6C!

It seems HackSplash won't work, since it only packs 2 JPG files into a raw body (in our case we need 6 GIF in a raw body), do you confirm it won't work?

> please use either UEFITool 0.28.0, 

Would you have the official download link for it? (I have found many websites on internet, and it's difficult to find the official trustable one). Do you think this version allows replacement of a GIF image?

> latest PhoenixTool for it.

Same problem here, there seems to be some non-trusted sources, such as the first link. Would you have the official link?

[image]

Thanks again!

NikolajSchlej, 2023-09-18:
Meant "firmware image". The official PhoenixTool source is [this MDL forum topic](https://forums.mydigitallife.net/threads/tool-to-insert-replace-slic-in-phoenix-insyde-dell-efi-bioses.13194/), needs a registration to see download links.

josephernest, 2023-09-18:
Thanks @NikolajSchlej. PhoenixTool will only allow to update a raw body (like UEFITool 0.28 does), but not change "easily" one logo image, is that correct? 

Or does PhoenixTool has a logo replacement feature included?

NikolajSchlej, 2023-09-26:
Correct, PhoenixTool can be used for generic UEFI file modification, and has nothing to do with logo replacement in particular. It's primary function is/was to insert/update/remove SLIC table(s) that were useful for Windows OEM activation, but it has a very robust UEFI file modification engine inside, which is in some ways better than UEFITool's.