LongSoft/UEFITool #378: Question but no discussion forum for these

kocoman1, 2023-09-24:
How to learn 

Vss store
Ftw store
Fdc store

How to restore cmos/Nv settings to default in insyde UEFI bios thx

NikolajSchlej, 2023-09-26:
VSS: https://github.com/tianocore/edk2/blob/bf0bdacdd6f6cdd2e9ac5db14b6daf19a5a5bd57/MdeModulePkg/Include/Guid/VariableFormat.h#L87
FTW: https://github.com/tianocore/edk2/blob/bf0bdacdd6f6cdd2e9ac5db14b6daf19a5a5bd57/MdeModulePkg/Include/Guid/SystemNvDataGuid.h#L62
FDC: was reverse-engineered, so I can't provide any references, but it's mostly a wrapper around VSS2 storage used for SecureBoot-related authenticated variables.

As for how to restore defaults, it depends on the particular image, so I can't provide a generic answer here. If you can upload an image you are trying to restore the defaults for, we can continue this conversation.

closed 2023-09-26 by NikolajSchlej

kocoman1, 2023-10-03:
How can I send you the bios image privately ? Thank you

NikolajSchlej, 2023-10-03:
You can email it to schlej[at]live.de.