LongSoft/UEFITool #379: Lenovo P520 BIOS file has DXE section compressed?

Electron-PSO, 2023-10-08:
Hello o/

I'm helping someone to add Resizable BAR to their mainboard.  I'm helping over Discord, which makes it a challenge.

I'm trying to perform the mod as per https://github.com/xCuri0/ReBarUEFI

The system in question is a Lenovo P520.  I have attached the rom file from the Lenovo updater.
[IMAGES03.zip](https://github.com/LongSoft/UEFITool/files/12841721/IMAGES03.zip)

I have opened the file with UEFITool to add the ffs module.  I am unable to see the usual DXE driver section.  Compression?

I have opened the file with MMTool and can see the DXE section.

Would it be better to work from a Firmware dump direct from the board?

I'm trying to avoid bricking the system for the user, as I don't think they are able to take the chip off and reflash it.

Your thoughts are appreciated,

Best regards

-68k

P.S   I've helped quite a few users to do this mod now.  I and they are very grateful for the UEFI tools.

NikolajSchlej, 2023-10-10:
It's actually not compression, but encapsulation into a BiosGuard (PFAT) image. Operating on an actual dump will help avoid this issue. 
There's also a way to remove PFAT stuff from this image by using (this script)[https://github.com/platomav/BIOSUtilities/blob/main/AMI_PFAT_Extract.py].

closed 2023-10-10 by NikolajSchlej

Electron-PSO, 2023-10-14:
Thankyou very much for your reply.  I'll look into it for the user.  

Getting a dump is proving difficult.  It may just be the user making mistakes or the tool tried wasn't up to the task.

I tend to just take the chip off the board and read it, but the user isn't able to do that.

I'll send the user links to other tools, maybe we can get a file from it.

Thanks again and best regards,

-E