LongSoft/UEFITool #383: are all the Micron SPI parts mis-identified?

WilliamLeara, 2023-10-31:
I recently submitted PR #382 with the intention of adding a new 128Mb SPI part.

After further review, it seems there may be a fundamental problem with the Micron section of descriptor.cpp.  Micron's vendor ID is assumed in the code to be 0x20, but I believe it is actually 0x2C.  Vendor ID 0x20 is XMC (Xinxin Semiconductor Manufacturing), a competitor SPI part vendor.

If you look at JEDEC's _JEP106BH STANDARD MANUFACTURERS IDENTIFICATION CODE_ spec, it says Micron Technology is 0x2C, and XMC is 0x20:  https://www.jedec.org/standards-documents/docs/jep-106ab
[image]
[image]

Going back to 2016 and SHA# cb430456 is looks like the Micron section was added assuming 0x20.  My hunch is that we need to change the vendor IDs of all the parts that begin with M/N/MT to 0x2C, and the parts that begin with XM to stay at 0x20, but put them into a new section of the file dedicated to XMC.

Any comment?  Does anyone have more background than me on this section of the code?

WilliamLeara, 2023-10-31:
Update:  did some investigation with a Dediprog ROM burner.  According to Dediprog's database:
[image]
and
[image]

So it appears that all the vendor and device IDs are correct, but that Micron and XMC share vendor ID code 0x20?  I don't get that, but my hunch now is that we need to break out the parts beginning with "XM" to a new section dedicated to XMC.  Besides that, the IDs appear to be correct.

Let me know if this plan sounds okay to you, @NikolajSchlej.

NikolajSchlej, 2023-10-31:
Sure, go ahead. Those JEDEC IDs are a mess, and most of them are obtained from various sources of questionable quality, so any improvements there are greatly appreciated. 
[Flashrom project had been a very solid source of them over the years](https://github.com/flashrom/flashrom/blob/master/include/flashchips.h).

NikolajSchlej, 2023-10-31:
```#define ST_ID			0x20	/* ST / SGS/Thomson / Numonyx / XMC(later acquired by Micron) */```
Good luck. :)

NikolajSchlej, 2023-11-02:
I'm closing this one because the answer to the original question is clearly "no, they are identified correctly, because Micron is now owning XMC, so calling those parts Micron makes sense".

closed 2023-11-02 by NikolajSchlej