LongSoft/UEFITool #385: Can not extract VBIOS and GOP

peter-held, 2023-12-03:
Hi,

I tried to extract VBIOS and GOP from this Intel NUC BIOS update and I couldn't find them.

Can you help me with it, please ?

You can download the BIOS file from here:
https://www.intel.com/content/www/us/en/download/773996/bios-update-anrpl357.html

Thanks.

NikolajSchlej, 2023-12-05:
The file is not an UEFI image, but an AMI PFAT capsule that needs to be extracted first using @platomav's excellent [BIOSUtilities](https://github.com/platomav/BIOSUtilities) toolset, namely [AMI_PFAT_Extract.py](https://github.com/platomav/BIOSUtilities/blob/main/AMI_PFAT_Extract.py) script.

Once extracted, the resulting *_ALL.bin file can be opened in UEFITool as usual, and will have both VBIOS and GOP visible in the tree.

[00 -- AN0027.CAP_ALL.bin.zip](https://github.com/LongSoft/UEFITool/files/13556369/00.--.AN0027.CAP_ALL.bin.zip)

closed 2023-12-05 by NikolajSchlej

peter-held, 2023-12-05:
I still can not find them.

After opening the .bin file in UEFITool,  I'm expecting to see something like here:

https://github.com/Kethen/edk2/blob/edk2-intel-gop/extract_vbt_gop.md

Which names should I look for ?

Thanks.

NikolajSchlej, 2023-12-05:
[image]
[image]
Haven't found the GOP driver, but it's most likely implemented as a combination of an Intel Graphics PEI driver and OemGop driver that just republishes the protocol.

NikolajSchlej, 2023-12-06:
Take whatever is closer to your real configuration.

No, you can't build a GOP driver for OVMF with what you have there because OVMF lacks a PEI phase. However, it will likely work to grab a GOP driver from BIOS for a similar board that supports same CPU modules, but doesn't use FSP.

peter-held, 2023-12-13:
I found a Dell laptop with the same processor
https://www.dell.com/support/home/en-in/product-support/product/vostro-16-5630-laptop/drivers
extracted the BIOS and GOP Driver.

Can you confirm me, please, that this is what it needs (that doesn't use FSP) ?
[gop.zip](https://github.com/LongSoft/UEFITool/files/13658224/gop.zip)

Thanks.

NikolajSchlej, 2023-12-13:
Yep, this is a traditional DXE GOP driver.