LongSoft/UEFITool #390: qt.qpa.plugin: Could not find the Qt platform plugin "xcb" in ""

Pandapip1, 2024-02-23:
When attempting to start `uefitool`, I get the following error:

```console
gavin@gavin-ideapad-1:~$ uefitool 
qt.qpa.plugin: Could not find the Qt platform plugin "xcb" in ""
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

Aborted
```

From my limited research it appears this is due to the fact that I am not running an X11 session (I am using Plasma Wayland instead).

NikolajSchlej, 2024-02-23:
Also, providing more details about your OS version and environment would be helpful.

Pandapip1, 2024-02-26:
The OS that I installed was elementary OS, but I've made so many customizations to it that basically all the evidence that's left is the apt sources, the app store, and the neofetch data.

~~Again, the problem is almost certainly that I'm using Wayland instead of X11.~~ Never mind, also fails to launch on XOrg.

Here's hopefully all the information on my system you could ever want:

```console
gavin@gavin-ideapad-1:~$ neofetch 
         eeeeeeeeeeeeeeeee            gavin@gavin-ideapad-1 
      eeeeeeeeeeeeeeeeeeeeeee         --------------------- 
    eeeee  eeeeeeeeeeee   eeeee       OS: elementary OS 7.1 Horus x86 
  eeee   eeeee       eee     eeee     Host: 82R4 IdeaPad 1 15ALC7 
 eeee   eeee          eee     eeee    Kernel: 6.5.0-18-generic 
eee    eee            eee       eee   Uptime: 19 mins 
eee   eee            eee        eee   Packages: 3569 (dpkg), 91 (flat 
ee    eee           eeee       eeee   Shell: bash 5.1.16 
ee    eee         eeeee      eeeeee   Resolution: 1920x1080 
ee    eee       eeeee      eeeee ee   DE: Plasma 5.24.7 
eee   eeee   eeeeee      eeeee  eee   WM: KWin 
eee    eeeeeeeeee     eeeeee    eee   WM Theme: Layan-solid 
 eeeeeeeeeeeeeeeeeeeeeeee    eeeee    Theme: [Plasma], Breeze [GTK2/3 
  eeeeeeee eeeeeeeeeeee      eeee     Icons: [Plasma], Fluent-dark [G 
    eeeee                 eeeee       Terminal: BlackBox 
      eeeeeee         eeeeeee         CPU: AMD Ryzen 7 5700U with Rad 
         eeeeeeeeeeeeeeeee            GPU: AMD ATI 04:00.0 Lucienne 
                                      Memory: 5017MiB / 9802MiB
gavin@gavin-ideapad-1:~$ sudo apt-get install '^libxcb.*-dev' libx11-xcb-dev libglu1-mesa-dev libxrender-dev libxi-dev libxkbcommon-dev libxkbcommon-x11-dev
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Note, selecting 'libxcb-sync-dev' for regex '^libxcb.*-dev'
Note, selecting 'libxcb-damage0-dev' for regex '^libxcb.*-dev'
Note, selecting 'libxcb-ewmh1-dev' for regex '^libxcb.*-dev'
Note, selecting 'libxcb-imdkit-dev' for regex '^libxcb.*-dev'
Note, selecting 'libxcb-present-dev' for regex '^libxcb.*-dev'
Note, selecting 'libxcb-xfixes0-dev' for regex '^libxcb.*-dev'
Note, selecting 'libxcb-xf86dri0-dev' for regex '^libxcb.*-dev'
Note, selecting 'libxcb-sync0-dev' for regex '^libxcb.*-dev'
Note, selecting 'libxcb1-dev' for regex '^libxcb.*-dev'
Note, selecting 'libxcb-composite0-dev' for regex '^libxcb.*-dev'
Note, selecting 'libxcb-xinerama0-dev' for regex '^libxcb.*-dev'
Note, selecting 'libxcb-xv0-dev' for regex '^libxcb.*-dev'
Note, selecting 'libxcb-dri3-dev' for regex '^libxcb.*-dev'
Note, selecting 'libxcb-xkb-dev' for regex '^libxcb.*-dev'
Note, selecting 'libxcb-image0-dev' for regex '^libxcb.*-dev'
Note, selecting 'libxcb-event1-dev' for regex '^libxcb.*-dev'
Note, selecting 'libxcb-dri2-0-dev' for regex '^libxcb.*-dev'
Note, selecting 'libxcb-util-dev' for regex '^libxcb.*-dev'
Note, selecting 'libxcb-shm0-dev' for regex '^libxcb.*-dev'
Note, selecting 'libxcb-res0-dev' for regex '^libxcb.*-dev'
Note, selecting 'libxcb-shape0-dev' for regex '^libxcb.*-dev'
Note, selecting 'libxcb-render-util0-dev' for regex '^libxcb.*-dev'
Note, selecting 'libxcb-xvmc0-dev' for regex '^libxcb.*-dev'
Note, selecting 'libxcb-ewmh-dev' for regex '^libxcb.*-dev'
Note, selecting 'libxcb-render0-dev' for regex '^libxcb.*-dev'
Note, selecting 'libxcb-cursor-dev' for regex '^libxcb.*-dev'
Note, selecting 'libxcb-randr0-dev' for regex '^libxcb.*-dev'
Note, selecting 'libxcb-xinput-dev' for regex '^libxcb.*-dev'
Note, selecting 'libxcb-dpms0-dev' for regex '^libxcb.*-dev'
Note, selecting 'libxcb-record0-dev' for regex '^libxcb.*-dev'
Note, selecting 'libxcb-xtest0-dev' for regex '^libxcb.*-dev'
Note, selecting 'libxcb-icccm1-dev' for regex '^libxcb.*-dev'
Note, selecting 'libxcb-reply1-dev' for regex '^libxcb.*-dev'
Note, selecting 'libxcb-property1-dev' for regex '^libxcb.*-dev'
Note, selecting 'libxcb-screensaver0-dev' for regex '^libxcb.*-dev'
Note, selecting 'libxcb-keysyms1-dev' for regex '^libxcb.*-dev'
Note, selecting 'libxcb-icccm4-dev' for regex '^libxcb.*-dev'
Note, selecting 'libxcb-aux0-dev' for regex '^libxcb.*-dev'
Note, selecting 'libxcb-util0-dev' for regex '^libxcb.*-dev'
Note, selecting 'libxcb-glx0-dev' for regex '^libxcb.*-dev'
[... 29 lines cut ...]
libxcb-xinput-dev is already the newest version (1.14-3ubuntu3).
libxcb-xkb-dev is already the newest version (1.14-3ubuntu3).
libxcb-xtest0-dev is already the newest version (1.14-3ubuntu3).
libxcb-xv0-dev is already the newest version (1.14-3ubuntu3).
libxcb-xvmc0-dev is already the newest version (1.14-3ubuntu3).
libxcb1-dev is already the newest version (1.14-3ubuntu3).
libxi-dev is already the newest version (2:1.8-1build1).
libxkbcommon-dev is already the newest version (1.4.0-1).
libxkbcommon-x11-dev is already the newest version (1.4.0-1).
libxrender-dev is already the newest version (1:0.9.10-1build4).
libxcb-cursor-dev is already the newest version (0.1.1-4ubuntu1).
libxcb-imdkit-dev is already the newest version (1.0.3-1).
libxcb-xrm-dev is already the newest version (1.0-3).
libx11-xcb-dev is already the newest version (2:1.7.5-1ubuntu0.3).
0 upgraded, 0 newly installed, 0 to remove and 41 not upgraded.
gavin@gavin-ideapad-1:~$ which uefitool 
/home/gavin/bin/uefitool
gavin@gavin-ideapad-1:~$ echo $XDG_SESSION_TYPE 
x11
gavin@gavin-ideapad-1:~$ uefitool 
qt.qpa.plugin: Could not find the Qt platform plugin "offscreen" in ""
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

Aborted
```

NikolajSchlej, 2024-02-27:
Now you have a different plugin failing, so it feels like there's something wrong with your Qt installation.
Can you run UEFITool with QT_DEBUG_PLUGINS=1: `QT_DEBUG_PLUGINS=1 uefitool` and share the output here?

NikolajSchlej, 2024-03-09:
Installed elOS 7.1 Horus on emulated x86-64 QEMU machine, then installed UEFITool from Flatpack like this:
- `flatpak remote-add --if-not-exists flathub https://dl.flathub.org/repo/flathub.flatpakrepo`
- `flatpack install uefitool`

It installed some additional components and the resulting app runs just fine.
[image]

closed 2024-03-09 by NikolajSchlej