LongSoft/UEFITool #393: Modded BIOS file fails to flash, "Selected file is not a proper BIOS!"

MrAnonymous7122, 2024-03-09:
Apologies if this is a silly issue, but I would like some help.

 I am trying to use [ReBAR](https://github.com/xCuri0/ReBarUEFI/wiki). I followed what the guide said exactly.

1. Get latest version of BIOS for your motherboard(I have Asus H110M-R BIOS Version 4210) 
2. Open BIOS File with [UEFITool](https://github.com/LongSoft/UEFITool/releases/tag/0.28.0) and add the ReBarDxe.ffs to the BIOS file.
3. Open the ReBarDxe BIOS file with [UEFIPatch](https://github.com/LongSoft/UEFITool/releases/tag/0.28.0) and use the [patches.txt](https://github.com/xCuri0/ReBarUEFI/blob/master/UEFIPatch/patches.txt)
4. Enable Above 4G Decoding and disable CSM
5. Flash BIOS and ReBAR should be working

However when I try to flash it to the motherboard using EZ Flash 3, it gives this error.
I'm not sure if it's worth noting that this motherboard needs a .cap file for BIOS.
I remember there being something about "Aptio signature may be invalid" when using UEFITool and UEFIPatch.
I have also made an issue on the ReBar repo but I was wondering if the problem is to do with UEFITool.
Thank you for spending your time to read and any help is appreciated.

NikolajSchlej, 2024-03-09:
Use a dedicated SPI programming hardware (CH341A-based programmers are cheap as dirt and fairly reliable), Asus likely finally implemented some verification routines for their EZ Flash software on the TPU/EC.

closed 2024-03-09 by NikolajSchlej

MrAnonymous7122, 2024-03-09:
Do you know if it's possible to use a Raspberry Pi as a programmer thing? I have one on hand but if not, I'll look at getting the CH341A thing.

NikolajSchlej, 2024-03-10:
It definitely is: https://winraid.level1techs.com/t/guide-recover-from-failed-bios-flash-using-raspberry-pi/30264

MrAnonymous7122, 2024-03-19:
Thanks for your help, the winraid guide was a bit out of date but more or less helpful and I have rebar working successfully.