LongSoft/UEFITool #394: Please help GUID defined section with unknown processing method Bla bla errors

zgrcn41, 2024-03-10:
parseImageFile: Aptio capsule signature may become invalid after image modifications
parseBios: one of volumes inside overlaps the end of data
parseBios: one of volumes inside overlaps the end of data
parseSection: GUID defined section with unknown processing method
parseSection: GUID defined section can not be processed
parseBios: one of volumes inside overlaps the end of data
parseBios: one of volumes inside overlaps the end of data
parseSection: GUID defined section with unknown processing method
parseSection: GUID defined section can not be processed

Errors

zgrcn41, 2024-03-11:
Nothing my mb ı think ım so tired finding double for uefi tool shit

NikolajSchlej, 2024-03-11:
Link to the bios?

closed 2024-03-11 by NikolajSchlej

zgrcn41, 2024-03-11:
https://dlcdnets.asus.com/pub/ASUS/mb/BIOS/PRIME-B450M-A-II-ASUS-4410.zip?model=PRIME%20B450M-A%20II

zgrcn41, 2024-03-11:
> Link to the bios?

https://dlcdnets.asus.com/pub/ASUS/mb/BIOS/PRIME-B450M-A-II-ASUS-4410.zip?model=PRIME%20B450M-A%20II

NikolajSchlej, 2024-03-11:
You can safely ignore all those messages, as they are informational, not errors. However, you can not ignore the very last one: "No patches can be applied to input file".
Your BIOS doesn't have the bytes that your patches.txt tries to patch, so there's nothing to patch in your BIOS.

zgrcn41, 2024-03-11:
> Hata değil, bilgi amaçlı oldukları için tüm bu mesajları güvenle göz ardı edebilirsiniz. Ancak sonuncuyu göz ardı edemezsiniz: "Giriş dosyasına hiçbir yama uygulanamaz". BIOS'unuzda patch.txt dosyanızın yamamaya çalıştığı baytlar bulunmadığından BIOS'unuzda yamalanacak hiçbir şey yoktur.

So what does this mean? Can't it be changed? Can't the BIOS be changed?

NikolajSchlej, 2024-03-12:
It means the patches you are trying to apply can not be applied. Need different set of patches developed for your BIOS. It's not changed because there is nothing to change in it.

zgrcn41, 2024-03-12:
Can you give me sets for my bios please 

> It means the patches you are trying to apply can not be applied. Need different set of patches developed for your BIOS. It's not changed because there is nothing to change in it.

NikolajSchlej, 2024-03-12:
You need to ask this question in ReBAR repo issues, UEFITool is not relevant here.