LongSoft/UEFITool #401: Invalid EnvVar's Attributes value for linked NVAR entry

john-8998, 2024-08-10:
When there's NVAR entry of subtype "Link" that link to a Data/Full NVAR entry, the link's attribute does not propagate to the linked NVAR.

NikolajSchlej, 2024-08-22:
Unclear if they should be, need to reverse the AMI implementation some more to see what attribute field is actually read by GetVariable.

john-8998, 2024-08-22:
Since I have an AMI UEFI, I don't need to reverse GetVariables(), I can just query the variables from the OS API.
When i read the variables in subject by Windows API, I get the attributes of the "Link" entry, not the Data/Full entry.

NikolajSchlej, 2024-08-24:
It's still unclear what change you are proposing. The information displayed by UEFITool is specific to the item that is selected, so if you select the link entry, you will get "correct" attributes, and if you select the Data/Full, you will get whatever that NVAR has. Do you expect something else?

john-8998, 2024-08-27:
You are right, that was what I've expected - that UEFITool will show the "raw" entry, whether it's "Link" or "Full".
But, as I understand from reading UEFITool's code that not the case, since the data of the "Full"/"Data" entry is propagated artificially to the "Link" entry.

NikolajSchlej, 2025-02-14:
I do not see any propagation happening here, but I do see that the attributes of the very first Link are the ones that define UEFI GetVariable call, and all other Links and the final Data entry will only have `Valid` and 'DataOnly'. I understand that this behavior might not be expected, but I stand with the original design and will close this issue as N/A.

closed 2025-02-14 by NikolajSchlej