LongSoft/UEFITool #404: UEFIFind cannot find text pattern

platomav, 2024-10-09:
Hello,

At the following Apple UEFI image, I am trying to detect the following pattern:

`244942494F534924................................2E00............2E00................2E00............2E00........................................0000`

This pattern should exist under two duplicate GUIDs "C3E36D09-8294-4B97-A857-D5288FE33E28".

[image]

Checking the raw body, I can see that it is indeed 0x4A (same size as the pattern) and I can trace it from the start to the end (two 0x00 bytes). However, UEFIFind cannot find it for some reason. If I remove one null byte at the end, it finds it, though:

[image]

Just a wild guess, could it be something new line related? Or am I doing something wrong?

[MBA11_00BB_03B_LOCKED.zip](https://github.com/user-attachments/files/17317648/MBA11_00BB_03B_LOCKED.zip)

vit9696, 2024-10-09:
I should have fixed this in master, please check that the latest build works fine.

platomav, 2024-10-09:
Yes, it works now! I downloaded the win64 artifact, and it produced the expected output:

[image]

Maybe this can be part of a release with a version increase so that other people can download it as a dependency.

Thanks for the quick fix @vit9696 💯

closed 2024-10-09 by platomav