LongSoft/UEFITool #406: UEFIFind fails to open input file when path is long on Windows

platomav, 2024-10-27:
Hello,

This ticket seems directly related to #363, but now the affected utility is UEFIFind and not UEFIExtract. Specifically, UEFIFind is not able to load a file when its path is "long" on Windows (MAX_PATH, UNC), failing with exit code 5 ([U_FILE_OPEN](https://github.com/LongSoft/UEFITool/blob/0030ea9481b1644ea6c00ba0050f35be6e75ad34/common/basetypes.h#L28C9-L28C44)).

```cmd
C:\Users\plato\Desktop>UEFIFind.exe "C:\Users\plato\Desktop\test_test_test_test_test_test_test_test_test_test_test_test\pkg\MacBookProEFIUpdate_2.pkg_extracted\temp\Payload_gzip_cpio\Payload~_extracted\System\Library\CoreServices\Firmware Updates\MacBookProEFIUpdateTwo-2.8\MBP51_007E_06B_LOCKED.scap" body list 244942494F534924 || echo %errorlevel%
5
```

Test Path - Success

``` text
C:\Users\plato\Desktop\test_test\pkg\MacBookProEFIUpdate_2.pkg_extracted\temp\Payload_gzip_cpio\Payload~_extracted\System\Library\CoreServices\Firmware Updates\MacBookProEFIUpdateTwo-2.8\MBP51_007E_06B_LOCKED.scap
```

Test Path - Failure

``` text
C:\Users\plato\Desktop\test_test_test_test_test_test_test_test_test_test_test_test\pkg\MacBookProEFIUpdate_2.pkg_extracted\temp\Payload_gzip_cpio\Payload~_extracted\System\Library\CoreServices\Firmware Updates\MacBookProEFIUpdateTwo-2.8\MBP51_007E_06B_LOCKED.scap
```

[test_long_path.zip](https://github.com/user-attachments/files/17533557/test_long_path.zip)

NikolajSchlej, 2025-02-12:
Should be fixed by https://github.com/LongSoft/UEFITool/commit/371448d0ecc97af3e0d564006edd77b9d227e2a8

NikolajSchlej, 2025-02-12:
Can't really test it due to lack of access to a Windows machine, but the fix is a direct port of UEFIExtract fix that had been proven to work before.

closed 2025-02-12 by NikolajSchlej

platomav, 2025-02-12:
Thank you Nikolaj, I can verify that it works:

[image]