LongSoft/UEFITool #414: Не запускаются pre-build бинарники на win11

nrBor, 2025-03-12:
При запуске любых pre-built binaries (https://github.com/LongSoft/UEFITool/releases) на win11 происходит просто вылет, хотя предыдущая версия uefitool норм работает. Это как-то связано с qt? Компилировать нет возможности. Что можно сделать?

NikolajSchlej, 2025-03-12:
У меня все работает в Windows 11
[image]

Нужны подробности: версия Windows 11, какая версия работает, какая версия не работает, и т.п.

nrBor, 2025-03-12:
Прошу прощения все действительно работает, нужно было снять галочку в свойствах - файл был получен с другого компьютера и заблокирован (первый раз такое вижу)

nrBor, 2025-03-12:
[image]

closed 2025-03-12 by NikolajSchlej