LongSoft/UEFITool #429: InsydeH2O BIOS firmware from Honor Magicbook Pro

mshcherbyna99, 2025-05-02:
Maybe it will be interesting for you all (maybe someone willing to help me at least find even hashed version of my password (not saying to find an actual password): [my repo](https://github.com/mshcherbyna99/redesigned-enigma/tree/main) containing 16MB of a stock firmware and locked dump.

NikolajSchlej, 2025-05-03:
Quick reports compare suggests that the password is stored somewhere in NVRAM, so clearing it will also clear the password.

To be able to pinpoint the exact password location, do the following: flash "unlocked" image, let the PC boot to BIOS Setup, set admin password from there, power off and make a new "locked" dump. This will guarantee that the only changes made would be the PW itself, and some minor ones happening every boot.

This is not an issue in UEFITool though, so I'm closing it now, but we can continue discussion in it, if needed.

closed 2025-05-03 by NikolajSchlej

mshcherbyna99, 2025-05-10:
Yeah, thank you very much for your help. A problem is that I have a bad connection to my chip and wanted to just find my password or at least a hash of it (SHA-256 or salted hash) in VSS storage of this dump. But as I said, thank you so much for help. 🙏