LongSoft/UEFITool #430: Legacy FV contents not parsed properly (size via header vs block)

platomav, 2025-05-04:
At the attached sample, FV `00504624-8A59-4EEB-BD0F-6B36E96128E0` contains another FV `7A9354D9-0468-444A-81CE-0BF617D890DF`, but its size (header vs block count) does not appear to be detected correctly.

[legacy_fv.zip](https://github.com/user-attachments/files/20012447/legacy_fv.zip)

More specifically, manually carving out that FV (header start to header start + header size) yields a FV of size 0x77000, but only 0x7000 is parsed when the source UEFI image is provided as is. This is possibly due to some weird legacy spec-breaking FV. Checking UEFITool, this is shown as "Non-UEFI data" followed by "Free Space" and "Padding":

[image]

Here is how that same FV is supposed to look like, once manually carved out (based on header-taken volume size):

[image]

Do you think there is there something that can be done to make these (spec-breaking, possibly) samples work without prior manual FV carving?

NikolajSchlej, 2025-05-05:
The issue here is that the outer volume has a size of 0x10000 both in the header and in the block map, so anything inside it can not be larger than that. This is why the internal volume is detected as being only 0x7000 in size instead of the correct value of 0x77000. 

We could come up with some heuristics on trying to detect cases like this, but as this is not a popular problem, and it a clear spec breakage, it is better to leave broken images broken, than to add yet another workaround that could easily backfire later for images that are truly broken/corrupted.

platomav, 2025-05-05:
Ah, you are right, I missed the parent volume declaring the wrong size.

Maybe it would be best to close this issue then, as it is a legacy case of a spec-breaking FV size, and it's not really possible to have workarounds for every broken UEFI-like blob out there.

Thank you for looking into this.